micropython/micropython #5178: Embedding Micropython Interpreter as FreeRTOS Task

Cubic-Rambo, 2019-10-03:
Hi everybody,
I am currently trying to embed Micropython in a bigger project based on an smt32f4 board.
Having set up FreeRTOS and some low level C task, I would like to place an high-level interpreter in a low priority task: so my choice of trying to include only Micropython as interpreter.
as I know, CC3200 port is base on FreeRTOS, and RT-Thread RTOS have been support micropython as a task.
First question: do you think it's possible to achieve this goal?Has anyone tried it?

jimmo, 2019-10-06:
> First question: do you think it's possible to achieve this goal?Has anyone tried it?

@Cubic-Rambo like you say, this is exactly how the ESP32 port works.

Have a look at esp32/main.c -- the important steps are:
 - create a freertos task
 - set it up as a micropython thread
 - create and initialise the micropython heap, and initalise the stack
 - initialise micropython (mp_init)
 - run python code or start a repl

The main thing to sort out then is hardware access and implementing Python APIs like any other port.

jonnor, 2024-07-27:
The question here seems to be well answered by @jimmo . Proposing to close this issue

closed 2024-07-28 by mattytrentini