micropython/micropython #5179: mpy-cross SIGSEGV on FreeBSD

decke, 2019-10-03:
```
# lldb ../../mpy-cross/mpy-cross
(lldb) target create "../../mpy-cross/mpy-cross"
Current executable set to '../../mpy-cross/mpy-cross' (x86_64).
(lldb) run -o build/frozen_mpy/upip_utarfile.mpy -s upip_utarfile.py -mcache-lookup-bc modules/upip_utarfile.py
Process 68769 launching
Process 68769 launched: '/root/ports/lang/micropython/work/micropython-1.11/mpy-cross/mpy-cross' (x86_64)
Process 68769 stopped
* thread #1, name = 'mpy-cross', stop reason = signal SIGSEGV: address access protected (fault address: 0x7fffffffe890)
    frame #0: 0x00007fffffffe890
->  0x7fffffffe890: adcb   %ch, %cl
(lldb) bt
* thread #1, name = 'mpy-cross', stop reason = signal SIGSEGV: address access protected (fault address: 0x7fffffffe890)
  * frame #0: 0x00007fffffffe890
    frame #1: 0x000000000027fc70 mpy-cross`compile_and_save(file="modules/upip_utarfile.py", output_file="build/frozen_mpy/upip_utarfile.mpy", source_file="upip_utarfile.py") at main.c:60:9
    frame #2: 0x000000000027f932 mpy-cross`main_(argc=7, argv=0x00007fffffffe9a8) at main.c:289:15
    frame #3: 0x000000000027fdf7 mpy-cross`main(argc=7, argv=0x00007fffffffe9a8) at main.c:304:12
    frame #4: 0x000000000021010f mpy-cross`_start(ap=<unavailable>, cleanup=<unavailable>) at crt1.c:76:7
```

mpy-cross segfaults when build on FreeBSD 12.0/amd64 both with gcc 9.2.0 and clang 8.0.1

The code for crt1.c is also on github:
https://github.com/freebsd/freebsd/blob/stable/12/lib/csu/amd64/crt1.c

Any idea what is going on here?

dpgeorge, 2019-10-04:
I managed to build an run both mpy-cross and the unix port of MicroPython on FreeBSD 12.0 amd64 using gcc 8.3.0.

decke, 2019-10-04:
It looks like we are doing some crap in the FreeBSD Portstree which adds additional flags and stuff. It builds fine (tested with gcc 8.3.0 and 9.2.0) when just unpacked and build manually so I'm sorry I've bothered you.

closed 2019-10-04 by decke