micropython/micropython #5181: __getattr__ and __setattr__ behaviors are now different from CPython 3.5

Arrowana, 2019-10-04:
We have some code that stopped working, but was perfectly functional on v1.9.4. The code was also unit tested on CPython. After upgrading to v1.11 it seems like the behavior of __setattr__ and __getattr__ changed.
I am running with v1.11 on a STM32F439.

The offending code, the idea is to provide an abstraction to be able to access attribute on _Configuration that would be proxied to the underlying dict 

`
class _ConfigurationTemplate:

	def __init__(self, configuration_name: str, configuration_data: dict):
		self._name = configuration_name
		self._data = configuration_data

class _Configuration(_ConfigurationTemplate):

	_BYPASS_SETATTR = ['_name', '_data', '_hash']

	def __init__(self, configuration_name: str, configuration_data: dict, configuration_hash: bytes):
		super().__init__(configuration_name, configuration_data)
		self._hash = configuration_hash

	def __getattr__(self, key):
		return self._data[key]

	def __setattr__(self, key, value):
		if key in self._BYPASS_SETATTR:
			super().__setattr__(key, value)
		else:
			self._data[key] = value

c = _Configuration('Yo', {'do': 1234}, None)
c.do
`

On v1.9.4 it works as intended and prints 1234

On v1.11 the constructor of _ConfigurationTemplate throws:
`Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "<stdin>", line 12, in __init__
  File "<stdin>", line 4, in __init__
  File "<stdin>", line 20, in __setattr__
AttributeError: 'super' object has no attribute '__setattr__'`

One work around I found is to implement __setattr__ in ConfigurationTemplate
def __setattr__(self, key, value):
    self.__dict__[key] = value

However, once this is introduced, another issue occurs on __getattr__
`Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "<stdin>", line 20, in __getattr__
< ---  More calls to __getattr__ ---->
  File "<stdin>", line 20, in __getattr__
RuntimeError: maximum recursion depth exceeded`

The behavior has changed because the test coverage did not actually cover how it use to work https://github.com/micropython/micropython/blob/b9ec6037edf5e6ff6f8f400d70f7351d1b0af67d/tests/basics/class_delattr_setattr.py#L36-L62

Is this a bug? This is at least introducing more differences between CPython and micropython

dpgeorge, 2019-10-04:
For the stm32 port, `__setattr__` was not enabled in v1.9.4.  It is enabled in v1.11.    So in v1.9.4 the getattr part would work (`c.do` would return 1234) but the setattr would not have been working.

dpgeorge, 2019-10-04:
I tested the above Configuration code (with `print(c.do)` added at the end) and indeed on the latest master version the stm32 and unix coverage builds (with setattr/delattr enabled) both fail.

So MicroPython's setattr is not 100% compliant with CPython, and this shows a case where it is different.

> Is this a bug? This is at least introducing more differences between CPython and micropython

In v1.9.4 setattr would be very different to CPython because it was not enabled on stm32.  Now that it's enabled the behaviour is closer to CPython, but, as found now, not perfect.

dpgeorge, 2019-10-04:
Adding some print's to both getattr and setattr shows the differences:

Output on CPython:
```
_Configuration.__setattr__ _name Yo
_Configuration.__setattr__ _data {'do': 1234}
_Configuration.__setattr__ _hash None
_Configuration.__getattr__ do
1234
```
Output on stm32 v1.9.4 (same as current unix port which has setattr/delattr disabled) (note that setattr is never called):
```
_Configuration.__getattr__ do
1234
```
Output on stm32 v1.11 (same as unix coverage build):
```
_Configuration.__setattr__ _name Yo
Traceback (most recent call last):
  File "<stdin>", line 26, in <module>
  File "<stdin>", line 12, in __init__
  File "<stdin>", line 4, in __init__
  File "<stdin>", line 22, in __setattr__
AttributeError: 'super' object has no attribute '__setattr__'
```

peterhinch, 2019-10-04:
See also [this forum thread](https://forum.micropython.org/viewtopic.php?f=2&t=7028).

Arrowana, 2019-10-07:
Thank you for the explanation, so should micropython behavior when setattr/delattr is enabled be modified to be closer to the CPython behavior?

dpgeorge, 2019-12-10:
This might be fixed by PR #5404