micropython/micropython #5183: It crashes if enable debug

xcguang, 2019-10-04:
https://github.com/micropython/micropython/pull/5182

dpgeorge, 2019-10-05:
Thanks for the report.  Can you please describe which port of MicroPython you see the error on, and where the GIL is being locked/unlocked before initialisation of the mutex?

xcguang, 2019-10-05:
I found it on esp32 platform, the crash information is
```
.[0;32mI (324) cpu_start: Pro cpu start user code.[0m
.[0;32mI (44) cpu_start: Chip Revision: 0.[0m
.[0;32mI (45) cpu_start: Starting scheduler on PRO CPU..[0m
.[0;32mI (0) cpu_start: Starting scheduler on APP CPU..[0m
/Users/xuchunguang/workspace/project/debug/micropython_gitee/ports/esp32/esp-idf/components/freertos/queue.c:719 (xQueueGenericSend)- assert failed!
abort() was called at PC 0x4008a080 on core 1

ELF file SHA256: 0000000000000000000000000000000000000000000000000000000000000000

Backtrace: 0x4008c7d7:0x3ffc21a0 0x4008caf5:0x3ffc21c0 0x4008a080:0x3ffc21e0 0x400f508e:0x3ffc2220 0x400f13fb:0x3ffc2240 0x400f1479:0x3ffc2260 0x400d46a1:0x3ffc2280 0x400d4b46:0x3ffc22a0 0x400f11df:0x3ffc2300 0x400d35bd:0x3ffc2350 0x400f1221:0x3ffc2370 0x40089b95:0x3ffc23a0

Rebooting...
ets Jun  8 2016 00:22:57
```

I traced the address:
```
xtensa-esp32-elf-addr2line -pfiaC -e build-GENERIC/application.elf 0x4008c7d7:0x3ffc21a0 0x4008caf5:0x3ffc21c0 0x4008a080:0x3ffc21e0 0x400f508e:0x3ffc2220 0x400f13fb:0x3ffc2240 0x400f1479:0x3ffc2260 0x400d46a1:0x3ffc2280 0x400d4b46:0x3ffc22a0 0x400f11df:0x3ffc2300 0x400d35bd:0x3ffc2350 0x400f1221:0x3ffc2370 0x40089b95:0x3ffc23a0
0x4008c7d7: invoke_abort at panic.c:?
0x4008caf5: abort at ??:?
0x4008a080: xQueueGenericSend at ??:?
0x400f508e: mp_thread_mutex_unlock at ??:?
0x400f13fb: mp_hal_stdout_tx_strn at ??:?
0x400f1479: mp_hal_stdout_tx_strn_cooked at ??:?
0x400d46a1: plat_print_strn at mpprint.c:?
0x400d4b46: mp_vprintf at ??:?
0x400f11df: DEBUG_printf at ??:?
0x400d35bd: gc_init at ??:?
0x400f1221: mp_task at ??:?
0x40089b95: vPortTaskWrapper at port.c:?
```

The reason of crash is  `DEBUG_printf ` called before mutex init.

dpgeorge, 2019-10-08:
> I found it on esp32 platform,

Ok, thanks, that makes sense.

agatti, 2024-09-26:
This was fixed in #15634, so it can be safely closed.

closed 2024-09-26 by xcguang