micropython/micropython #5186: BLE: Outstanding work

jimmo, 2019-10-06:
- [ ] ESP32 support (#5171)
 - [ ] Customisation of scan interval and window.
 - [ ] Documentation & sample code (in progress)
 - [ ] Pairing & bonding support.
 - [ ] NRF5x support
 - [ ] Configuration of encryption for characteristics. (ref: #5174)
 - [ ] Support for accessing "scan response" data (active scan)
 - [ ] REPL over BLE NUS.

peterhinch, 2019-10-06:
As a total BLE n00b I'm looking forward to the docs and samples :)

jonsmirl, 2019-10-06:
How about Unix BT support? I'd like to have that so that I can use my desktop for development. Of course Unix BT is way different but that would be hidden behind the API.

dmazzella, 2019-10-06:
Is BLE support for STM32WB also planned?

dpgeorge, 2019-10-08:
> Extract bytes from UUID class (buffer protocol?) for appending to adv_data.

Should be easiest and most flexible to use the buffer protocol, then `bytes(uuid)` would give the bytes value.

dpgeorge, 2019-10-08:
> How about Unix BLE support?

This would be good but no progress has been made yet.

> Is BLE support for STM32WB also planned?

Yes, but no timeline yet.

jonsmirl, 2019-10-09:
From the PR request thread...

ble.gap_advertise(100,'Micropython')
You need to pass in an advertising packet, eg `b'\x02\x01\x02\x04\x09MPY'

There needs to a helper API for constructing advertising packets. Most people are not familiar with the BLE spec.

jonsmirl, 2019-10-14:
I've been poking around looking at Unix support for Bluetooth. The easiest strategy looks to be using DBUS. Bluez always builds with dbus support so there is no getting rid of it. Buildroot contains the package 'dbus-python', http://dbus.freedesktop.org/releases/dbus-python  So can micropython access this package? If it can, then an approach like Adafruit used should work. https://github.com/adafruit/Adafruit_Python_BluefruitLE/blob/master/Adafruit_BluefruitLE/bluez_dbus/provider.py

dpgeorge, 2019-10-14:
> Is BLE support for STM32WB also planned?

See #5212 for a start on this.

dpgeorge, 2019-10-14:
> I've been poking around looking at Unix support for Bluetooth. The easiest strategy looks to be using DBUS.
> ...
> Does not appear to be simple to get dbus-python working in micropython. Maybe python expert can comment on best way to achieve this?
> 
> Looks like an attempt to implement a moddbus.c is here

Thanks for looking into this, definitely a good thing to work on.

If Bluez is the way to go, then there are probably a few ways to interface with it:
1. using system calls to run `bluetoothctl`
2. using the `ffi` module to interface to `libdbus.so`
3. implementing a new C module to interface to dbus, like `moddbus.c`

2 and 3 are basically the same, just 2 is implementing it in pure Python, 3 in C.

jonsmirl, 2019-10-14:
One goal of mine is to have API compatibility between my desktop and target systems (ESP32, Unix).  Desktop is running bluez and lots of stuff depends on that, so I don't see any obvious way to avoid it.  Another reason for keeping Bluez is if the target system is implementing Bluetooth audio.

I have not used Python much, which is a better approach, ffi or C module? Has anyone already coded micropython DBUS support using ffi?

All of this Bluez, dbus, glib, etc support is fairly large so I could also see someone porting code over from a microcontroller to directly access the BLE device. But doing that will lose the ability to run other Linux apps. I'm sure there are boards around running Linux that only have BLE hardware with custom drivers and no BlueZ. Of course these variations will be hidden below the common API.

ricou73, 2019-10-15:
currently i do some test on esp32-20191011-v1.11-422-g98c2eabaf.bin with an esp32-wroom.
My python code is very simple, unfortunatly this test show ony 2 ble device where my phone or a lopy4 found 6 ble device. Any clue about why scanning seem so poor on result ? (if i test in my office it's worst, showing only 5 device where nrf connect found 20 of them)

```
import bluetooth
import ubinascii
import time
bt = bluetooth.BLE()
bt.active(True)
devlist=list()
def bt_irq(evt,data):
	global devlist
	if(evt==16) :
		if not (str(ubinascii.hexlify(data[1])) in devlist):
			devlist.append(str(ubinascii.hexlify(data[1])))
	else:
		print("Evt:"+str(evt)+" DATA:"+str(data))
		EndOf=True
bt.irq(bt_irq)
bt.gap_scan(60000)
time.sleep(60)
print("End of scan")
print(devlist)
```

dmartauz, 2019-10-15:
Try to specify scan interval and window - using the same value will start continuous scanning, e.g.:
`gap_scan(60000, 30000, 30000)`. Interval and window values are in microseconds.

ricou73, 2019-10-15:
I guess i have to wait the next available binary build because on g98c2eabaf tag i have this error calling gap_scan
`TypeError: function expected at most 2 arguments, got 4`
(i saw the modification about gap_scan is commited on the 11 october). I will keep you informed

dmartauz, 2019-10-15:
You can build the binary yourself:
https://github.com/micropython/micropython/blob/master/ports/esp32/README.md

jonsmirl, 2019-10-16:
Another way to do Linux support is to access the functions via a socket like this code from the Bluez source illustrates, it is a test program.   ./configure --enable-testing  will build it.  The test program creates a peripheral device I was able to access on my phone.
https://git.kernel.org/pub/scm/bluetooth/bluez.git/tree/peripheral
As a bonus I believe this code can run either with or without the bluez stack running. This has potential to be much more memory efficient than the dbus approach since you can run it standalone is small systems.

ricou73, 2019-10-16:
> 
> 
> You can build the binary yourself:
> https://github.com/micropython/micropython/blob/master/ports/esp32/README.md

I have tryed but miserably failed. I have and windows with msys2, xtensia and was able to build lopy4 custom firmware, but when i try micropython build i run into a lot of trouble.
the last was when i run make i have the following error (before that i have patched pyserial to successfull ompilation for esptool)

CC ../../extmod/modbtree.c
In file included from ../../extmod/modbtree.c:37:0:
../../lib/berkeley-db-1.xx/PORT/include/db.h:1:1: error: expected identifier or '(' before '.' token
 ../../include/db.h
 ^
In file included from ../../extmod/modbtree.c:38:0:
../../lib/berkeley-db-1.xx/PORT/include/../../btree/btree.h:89:3: error: data definition has no type or storage class [-Werror]
 } PAGE;
   ^
../../lib/berkeley-db-1.xx/PORT/include/../../btree/btree.h:89:3: error: type defaults to 'int' in declaration of 'PAGE' [-Werror=implicit-int]
../../lib/berkeley-db-1.xx/PORT/include/../../btree/btree.h:126:2: error: unknown type name 'u_int32_t'
  u_int32_t ksize;  /* key size */
  ^

I have no idea to what i have to do for to correcting this, .

jonsmirl, 2019-10-16:
The littlevgl port should work for esp32 and BT. Set it to use the IDF 4 commit tag it will list for you.
https://github.com/littlevgl/lv_micropython

diginfo, 2019-10-18:
Is this a bug ??

```
import ubinascii
import bluetooth as bt
addr=ubinascii.unhexlify(b'cdd751150f7e')
bt.BLE.gap_connect(0,addr)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: argument has wrong type
```

jonsmirl, 2019-10-18:
New idea on how to do Unix support...  How about writing a transport driver for the Nimble Host piece that calls the Linux kernel BT API? Both of these APIs are HCI based. The Linux side needs to go through the kernel so that it won't disrupt other uses of the adapter like Bluetooth Audio. 

I am not familiar enough yet with the two APIs to understand what is involved. Anyone have more experience in this area?  One issue is that both stacks implement L2CAP.

dpgeorge, 2019-10-21:
>  How about writing a transport driver for the Nimble Host piece that calls the Linux kernel BT API?
> ...
> Nimble people informed me it has already been ported.

Yes, this would be a good solution because 1) it's the simplest way to get the `bluetooth` module working on Linux; 2) it will provide a way to test our nimble bindings.

> The downside is that it has to have exclusive use of the bluetooth device.

Right, so it might also be a good idea to provide a dbus/other solution as an alternative on Linux.

jonsmirl, 2019-10-21:
Quick solution for desktop is to just plug in an extra USB BT adapter. BlueZ can have one and micropython development can use the other.

mirko, 2019-10-30:
I was playing around with adding the flag `0x04` (=WRITE_WITHOUT_RESPONSE) (which then would be `0x08 | 0x04 = 0x12` (=WRITE+WRITE_WITHOUT_RESPONSE)) to the RX characteristics properties, hoping for faster transfers.
However transfers being received on the ESP32 running micropython appear corrupted.

diginfo, 2019-11-07:
Can you please add the ability to:

1) Get a list of connection handles
2) Get the RSSI of a connected device

Many Thanks

dmazzella, 2019-11-11:
@jimmo @dpgeorge some progress on

> - [ ] Pairing & bonding support.

?

Cubic-Rambo, 2019-11-13:
How to make the nRF52 board support the ubluetooth library？

playxz, 2019-11-20:
> @jimmo @dpgeorge some progress on
> 
> > * [ ]  Pairing & bonding support.
> 
> ?

Any news?

playxz, 2019-11-20:
What about descriptors?
How can I create them?
May I have some examples?

Thx

pacmac, 2019-12-21:
Is  "Get the RSSI of a connected device" anywhere in the pipeline :-)

dmazzella, 2020-01-22:
> @jimmo @dpgeorge some progress on
> 
> 
>  Pairing & bonding support.
> 
> 
> ?

news on this?

FynnHR, 2020-02-06:
Does pairing and bonding support also includes Passkey authorization?

pacmac, 2020-02-18:
Is "Get the RSSI of a connected device" anywhere in the pipeline ??

This would be useful :-)

Walkline80, 2020-03-25:
Waiting for pairing & bonding support

dpgeorge, 2020-04-29:
> I've been poking around looking at Unix support for Bluetooth.

The bluetooth module is now supported on unix builds, via btstack.  See 7563d582100d7947eb61960c9116c8fc10f1e996

Walkline80, 2020-05-08:
According to [NimBLE Host GAP Reference](https://mynewt.apache.org/latest/network/ble_hs/ble_gap.html#c.ble_gap_security_initiate), I'm trying to add pairing & bonding function, and its worked for me, but there's no way to auto reconnect....

Still waiting and hoping for pairing & bonding support

dmazzella, 2020-10-17:
Updates on pairing & bonding support?

street-grease-coder, 2021-01-05:
bump @pairing/bonding

jimmo, 2021-01-05:
> Updates on pairing & bonding support?
> bump @pairing/bonding

@dmazzella @street-grease-coder 

This was implemented in #6662 and will be in the 1.14 release for STM32 (e.g. PYBD and WB55).

However it's not supported on ESP32 due to the need for synchronous event handlers. Looking for a volunteer to sort out the necessary plumbing on ESP32. Se notes in #6594 and #6651

kt-work, 2021-02-23:
Is there any chance to get an event on completed advertisment transmission, that can be evaluated by the handler that has been  passed to BLE.irq()?

jimmo, 2021-02-24:
> Is there any chance to get an event on completed advertisment transmission, that can be evaluated by the handler that has been passed to BLE.irq()?

@kt-work I'm not sure whether this is supported by either NimBLE or BlueKitchen (or any HCI controller?)... can you provide a link to documentation for another BLE implementation that supports this and I can look into it.

I'm guessing you're looking at implementing some sort of data-over-advertising (e.g. mesh?). I had a look at NimBLE's mesh implementation and it works by starting advertising and sleeping for a number of intervals (i.e. it will always send more than one beacon).

kt-work, 2021-02-24:
Hi Jim,

I'm trying to introduce a sequence number into advertisements for an end of
line test.

In the past I did work on a meshing implementation based on a feature like
this. The device was a BCM20736. Broadcom/Cypress/Infineon provides an SDK
called WICED SDK that contains an application framework lets call it, that
supports a callback for finished advertisements.

It is quite possible that it will be impossible to implement if it is not
supported on the HCI layer of the STM32WB55.

I also considered starting advertising and stopping it to modify the next
advertisement. This might work if I can be certain that advertising starts
right after the call to start advertising or after a fixed amount of time.
If you could look into that it would help me too.

Best Regards,

Kilian Timmler

-- 
-- 
Kilian Timmler, Embedded Development

Tel: +49 (0) 351 - 219 71 444
Mobil: +49 (0) 151 - 28793879
Email: kilian.timmler@exelonix.com
Internet: http://www.exelonix.com/

exelonix GmbH
Washingtonstraße 16/16a
01139 Dresden
Germany

Geschäftsführer: Dr. Thorsten Dräger, Dr. Matthias Stege
Amtsgericht Dresden HRB 32275, Sitz der Gesellschaft: Dresden

martinlombana, 2021-03-08:
Hi! @jimmo, when can we expect to have bonding capabilities for the ESP32 port? Thanks!

street-grease-coder, 2021-03-08:
> Hi! @jimmo, when can we expect to have bonding capabilities for the ESP32 port? Thanks!

Hey, I don't think this is how open source works. They're already doing an amazing job. The way to accelerate this is by contributing, not by asking for deadlines that people can't keep because many are doing it in their free time and the process is usually nonlinear

martinlombana, 2021-03-08:
> Hey, I don't think this is how open source works. They're already doing an amazing job. The way to accelerate this is by contributing, not by asking for deadlines that people can't keep because many [...]

Hey yourself! Yes, That is why I supported @jimmo and started paying 3 days ago. Using micropython, asking questions in order to develope my knoledge further, even though I am very new in micropython and low-level stuff. 

I am not pressuring anyone. As a matter of fact my question was *just a question*. You might have imprinted a tone that was not even there to begin with. I re-read my question 10 times, with an open mind, in case there was something I was not seing. It was just a request for information, as I know he is actively developing A LOT of things, for which I am very grateful. Not pressuring anyone here. I am not like that.

Cheers,

street-grease-coder, 2021-03-08:
You should add how future readers can support Jimmo, I'd be in on that :) The rest is not my place to say - and further I'm also a newbie here

oclyke, 2021-06-22:
@jimmo I see the latest request for volunteers to work on synchronous events in the ESP32 port to support pairing/bonding. Have there been any other volunteers yet? Is there a dedicated thread/issue that we could use to coordinate our efforts?

My experience with BLE is limited, but I previously used the ESP IDF hal to implement pairing / bonding so perhaps that would be a good place to start. Please let me know how best I can help with this particular issue. Thanks!

leycec, 2021-12-08:
We'd all *love* to see bonding-on-ESP32, of course – but there appear to be [soft blockers in the ESP32 SDK API that sadly complicate this](https://github.com/micropython/micropython/issues/5186#issuecomment-754395320):

> However it's not supported on ESP32 due to the **need for synchronous event handlers.** Looking for a volunteer to sort out the necessary plumbing on ESP32. See notes in #6594 and #6651

Ideally, Espressif themselves would either financially sponsor *or* supply in-house PR manpower for this community effort that economically benefits them. That isn't happening, so... here we are.

Meanwhile, I'm just delighted that the [workhorse AIOBLE framework](https://github.com/micropython/micropython-lib/tree/master/micropython/bluetooth/aioble) continues to receive timely commits. Keep up all the incumbent greatness, @jimmo. You're a living embedded systems legend.

leycec, 2021-12-31:
**NRF5x BLE support** would be phenomenal as well.

Indeed, this would probably make a great first issue for someone with enthusiasm! <sup>*so, not me*</sup> Why? Because Adafruit have already done the hard work by adding full support for BLE (including pairing and bonding, crucially) to CircuitPython's existing NRF5x port. Unsurprisingly, [the `Adapter.c` and `Connection.c` files in the low-level `_bleio` C extension subdirectory of that port](https://github.com/adafruit/circuitpython/tree/main/ports/nrf/common-hal/_bleio) are where all of the good stuff pertaining to pairing happen; likewise, [the `bonding.c` file manages bonding](https://github.com/adafruit/circuitpython/blob/main/ports/nrf/common-hal/_bleio/bonding.c).

So that's a mostly solved problem. Phew! :sweat_smile:

puppet13th, 2023-06-27:
@dpgeorge @jimmo any new about implementing BLE.config() to set ble txpower. on wemos c3 mini v1.0, ble was unuseable because of antenna/hardware design. The wifi part can be [workaround](https://www.wemos.cc/en/latest/c3/c3_mini_1_0_0.html) using lower txpower.
I have tried to hardcoded ble txpower using these [instruction](https://forum.micropython.org/viewtopic.php?t=7733) from the old forum and it works well. BLE connection was stable, but i have not test impact on the range though.

ble scan result comparison:
```
DEVICE1	-40	lolin_c3_mini_stock
DEVICE1	-48	lolin_c3_mini_0dbm
DEVICE1	-55	ai_thinker_c3
DEVICE1	-56	ai_thinker_c3_3dbm
DEVICE1	-60	lolin_c3_mini_3dbm
		
DEVICE2	-49	lolin_c3_mini_stock
DEVICE2	-57	ai_thinker_c3
DEVICE2	-58	ai_thinker_c3_3dbm
DEVICE2	-60	lolin_c3_mini_3dbm
DEVICE2	-61	lolin_c3_mini_0dbm
		
DEVICE3	-39	lolin_c3_mini_0dbm
DEVICE3	-42	ai_thinker_c3
DEVICE3	-50	lolin_c3_mini_3dbm
DEVICE3	-54	ai_thinker_c3_3dbm
DEVICE3	-59	lolin_c3_mini_stock
		
DEVICE4	-62	lolin_c3_mini_stock
DEVICE4	-64	lolin_c3_mini_0dbm
DEVICE4	-66	lolin_c3_mini_3dbm
DEVICE4	-74	ai_thinker_c3
DEVICE4	-76	ai_thinker_c3_3dbm
```