micropython/micropython #5187: usocket.getaddrinfo() returns a blank list

vubon, 2019-10-06:
Getting blank list to get address
```python
MicroPython v1.11-381-g4102320e9 on 2019-10-02; ESP32 module with ESP32
Type "help()" for more information.
>>> import usocket
>>> usocket.getaddrinfo("google.com", 80)
[]
```

robert-hh, 2019-10-06:
No problem here. Just to ask a dumb question: is your ESP32 connected to the Internet?

vubon, 2019-10-06:
hmm. Maybe or not, but the internet connection is okay.
```python
import time
import network
import socket
import usocket

connection = network.WLAN(network.STA_IF)
connection.active(True)
connection.connect("<SSID>", "<password>")
while not connection.isconnected():
    time.sleep(0.10)
    connection.connect("<SSID>", "<password>")

print(connection.isconnected())
print("usocket", usocket.getaddrinfo("google.com", 80))
print("socket", socket.getaddrinfo("vubon.me", 443))
```
Output:
```python
True  # means the internet connection is okay 
usocket [] # no result 
socket [] # no result 
```
Could I miss anything here ?? Kindly let me know if I missed anyting . Thank you for your reply

vubon, 2019-10-06:
Fixed the problem. I need to change my wifi connection. Thank you

closed 2019-10-06 by vubon

robert-hh, 2019-10-06:
You do not have to repeat the `connection.connect("<SSID>", "<password>")` in the while loop. Just issue it once and wait for the connection. I understand, taht you connect to your local router, and that this router allows access to the internet.