micropython/micropython #5194: micropython.org is down

mcauser, 2019-10-09:
https://downforeveryoneorjustme.com/micropython.org

> Is micropython.org down right now?
> It's not just you! micropython.org is down.

```
$ host micropython.org
micropython.org has address 176.58.119.26
micropython.org mail is handled by 10 micropython.org.
Mikes-MBP-3:micropython mike$ host -a micropython.org
Trying "micropython.org"
;; ->>HEADER<<- opcode: QUERY, status: NOERROR, id: 64852
;; flags: qr rd ra; QUERY: 1, ANSWER: 7, AUTHORITY: 0, ADDITIONAL: 0

;; QUESTION SECTION:
;micropython.org.		IN	ANY

;; ANSWER SECTION:
micropython.org.	14399	IN	A	176.58.119.26
micropython.org.	14399	IN	NS	ns1.dreamhost.com.
micropython.org.	14399	IN	NS	ns2.dreamhost.com.
micropython.org.	14399	IN	NS	ns3.dreamhost.com.
micropython.org.	14399	IN	SOA	ns1.dreamhost.com. hostmaster.dreamhost.com. 2016101200 20047 1800 1814400 14400
micropython.org.	14399	IN	MX	10 micropython.org.
micropython.org.	14399	IN	TXT	"v=spf1 a -all"

Received 205 bytes from 8.8.8.8#53 in 742 ms
```

```
$ ping 176.58.119.26
PING 176.58.119.26 (176.58.119.26): 56 data bytes
Request timeout for icmp_seq 0
Request timeout for icmp_seq 1
Request timeout for icmp_seq 2
```

mcauser, 2019-10-09:
> Is micropython.org down right now?
> It's just you. micropython.org is up.

closed 2019-10-09 by mcauser

MrSurly, 2019-10-09:
It's up for me.

dpgeorge, 2019-10-09:
It was down twice for a few minutes at a time to resize storage disks.

mcauser, 2019-10-09:
Right as I clicked `[submit]` on a forum post reply :)

dpgeorge, 2019-10-09:
> Right as I clicked `[submit]` on a forum post reply :)

Oh!  I think you can just "go back" in the browser to get the post text, it shouldn't be lost.