micropython/micropython #5196: [I2C] parameter keyword not accepted

rolandvs, 2019-10-09:
It seems that the “machine.I2C().writeto” and “.readfrom” accept a third parameter, however only the value and not the documented keyword like “stop=True” is accepted.

dpgeorge, 2019-10-09:
> however only the value and not the documented keyword like “stop=True” is accepted

That's correct, it's not a keyword argument, just a positional argument.  It's like this for a lot of built-in functions/methods, that you cannot specify the keyword.  This is to save code space.

Python actually has a way to denote such a thing, using forward slash, eg `i2c.writeto(addr, buf, stop=True, /)`.  And this will soon become an actual language/syntax feature: https://www.python.org/dev/peps/pep-0570/

So, I propose that the documentation is updated to use this notation, to indicate positional-only args.

dpgeorge, 2020-01-06:
Fixed by 99ed431d2840b30f67eae49be96d31c7faeca75e

closed 2020-01-06 by dpgeorge