micropython/micropython #5199: machine.RTC.memory undocumented and does not survive watchdog reset

goatchurchprime, 2019-10-09:
The [documentation for machine.RTC](http://docs.micropython.org/en/latest/library/machine.RTC.html) contains 7 methods, yet  yet `MicroPython v1.11-406-g4a6974bea on 2019-10-06; ESP32 module with ESP32` has only 3 methods: init(), datetime(), and memory().  

I assume the memory() function is supposed to access some non-volatile storage.  

```
>>> import machine; 
>>> r = machine.RTC(); 
>>> r.memory("hi there")

Cntl-D soft reboot
[reset ensues]

>>> import machine; 
>>> r = machine.RTC(); 
>>> print(machine.reset_cause(), r.memory())
5, b'hi there'
>>> machine.deepsleep(1000)

[reset ensues]
 
>>> import machine; 
>>> r = machine.RTC(); 
>>> print(machine.reset_cause(), r.memory())
4, b'hi there'
>>> machine.WDT(0, 1000)

[reset ensues]

>>> import machine; 
>>> r = machine.RTC(); 
>>> print(machine.reset_cause(), r.memory())
5, b''
```

Calling `machine.reset()` also wipes this RTK.memory storage.  

I guess we've found a way to tell the difference between the Ctrl-D soft reboot and a watchdog reset, outlined in issue #5134 (since the latter disappears the memory), but it's not quite what I want.  

The use of this storage is to enable some form of debugging wherein the code keeps updating where it is in the various suspect loops and functions, so that when the watchdog resets it you have a hope of finding out where it hung.

ajlennon, 2019-10-09:
I think this is causing your problem 

https://github.com/espressif/esp-idf/blob/master/components/esp32/cpu_start.c#L159

goatchurchprime, 2019-10-11:
The RTC.memory also gets cleared when the brownout detector was triggered (which also results in a machine.reset_cause() == 5 == machine.SOFT_RESET).

This means I can't use it to count the brownout resets.  Sometimes my ESP32 boards get into a state where their voltage regulators seem to get tired and there is a brownout reset every time it boots and executes `network.WLAN.activate(1)`.  This means it can sit in this loop repeatedly for hours until there is a lucky moment when it comes alive and can make a connection out.  

I have seen this repeated brownout on WLAN.activate behavior when connected (and powered) from a laptop USB so I can read the serial line.  But obviously when I wire it up to its own power supply and it goes into a locked room I can't be sure this is what's happening when it goes off-line for 6 hours a week later.  However, if when it came online it reported to me that it had experienced 4000 brownouts in a row, then knowing that each brownout-reboot-brownout cycle took 5 seconds, I'd have a provable diagnosis.

ajlennon, 2019-10-11:
There is already a magic number in there which is checked. It shouldn't need to be cleared down unless the magic number is invalid imho.

ajlennon, 2019-10-11:
@goatchurchprime did you look into whether the brownout trigger levels are configurable?

ajlennon, 2019-10-11:
Interesting - can you try this @goatchurchprime and let me know

https://arduino.stackexchange.com/questions/55702/esp32-disable-brownout-detector

peterhinch, 2019-10-12:
> ESP32 boards get into a state where their voltage regulators seem to get tired and there is a brownout reset every time it boots

In the interests of reliability I'd want to know the (electrical) cause of this. Could the regulator be overheating and shutting down until it has cooled? Is the behaviour dependent on the voltage powering the device?

goatchurchprime, 2019-10-12:
Me too.  The it stopped doing it once I took it down out of the ceiling where it was running a dotmatrix sign and left it on my desk for a day.  These long-delayed hardware issues are a real headache.  

What would help a lot, as well as being able to distinguish and count brownouts, would be a register that stored the time.ticks_ms() value from before the last reset so one could tell where in the bootup process it all went wrong.

ajlennon, 2019-10-12:
Should be trivial to patch the code.

Your problem is your beta test environment is significantly different from your alpha (bench) test environment which is slowing down your iterations to fix this.

ajlennon, 2019-10-12:
I don’t mind doing the work and submitting a PR but I don’t want to waste my time if it’s not accepted.

What’s the view from the maintainers I’m removing the wake reason check and leaving it to the magic number?

ajlennon, 2019-10-13:
Test build generic ESP32 binary here to try

https://github.com/DynamicDevices/micropython/releases/tag/v1.11-rtcbss

heikokue, 2020-12-20:
Hello,
I just pushed my proposal to describe the undocumented method machine.RTC.memory.
I hope, you'll enjoy it ;-)
See pull request #6713

karfas, 2021-05-22:
I added PR #7298 , which fixes most likely the "does not survive watchdog reset" bit of this issue.

mattytrentini, 2023-09-20:
(Just to summarise the status, particularly for @dpgeorge, since it's been a while!)

This ticket covers a couple of issues that are required to make `machine.RTC().memory()` usable for the esp32 port. Namely:

1. `memory()` is not documented
2. The memory is not always preserved, particularly over watchdog resets

If #6713 can be tidied up it looks appropriate to resolve 1. #7298 appears to resolve 2.

It would be good to sort these out soon.

Also, tangentially related, #7133 provides `RTC().usermem()` that allows `ctype` structures to be easily stored in RTC memory. I think it can be considered separate to 1 & 2 above, though there is some cross-over. 

Finally, esp32/esp8266 are currently the only ports that supports `RTC().memory()` - other ports _could_ provide it. And there are inconsistencies across ports with the other `RTC()` methods (especially `alarm`, `cancel` etc). I'll raise a separate ticket that summarises these wider `machine.RTC()` issues.

jonnor, 2024-03-27:
I now see documentation for RTC.memory at https://docs.micropython.org/en/latest/library/machine.RTC.html#machine.RTC.memory
and https://github.com/micropython/micropython/pull/7298 has been merged - and released as part of 1.22.0
Commit 78b3fe56897d029f2786256d83dd2fecacb34637

Proposing to close this. CC @mattytrentini

closed 2024-03-27 by mattytrentini