micropython/micropython #5200: esp32: Depreciated KConfig options causing broken builds with IDF 4.x

nevercast, 2019-10-09:
As of this commit https://github.com/espressif/esp-idf/commit/0ae53691bae8b841d35ed0d5a4d43e1f1a873d2a several KConfig options have been renamed, in the process several (my count, 17) rename files have been added, like this one: https://github.com/espressif/esp-idf/blob/master/components/esp_common/sdkconfig.rename which shows the new names for the options.

When building ESP32 with IDF 4.x, for example when trying the BLE functionality that has just landed, the sdkconfig files provided in micropython do not set the correct properties. For example:

https://github.com/micropython/micropython/blob/master/ports/esp32/boards/sdkconfig.base

This base config sets `CONFIG_TASK_WDT_CHECK_IDLE_TASK_CPU0` rather than `CONFIG_ESP_TASK_WDT_CHECK_IDLE_TASK_CPU0` which cases the IDF 4.x builds to fail with a watchdog reset every 5 seconds.

This issue is to propose solutions, my current thoughts are:

* Add the new names with the old config names and essentially duplicate the lines in our sdkconfigs 
* **OR** Find and replace the config names using the `sdkconfig.rename` files at build time when building the IDF 4 port, keeping the old names in the files.

Feel free to suggest other solutions.

dpgeorge, 2019-10-10:
Thanks for the report, I can confirm the issue (although the WDT timeout only shows up if the MCU does work, like an infinite loop).

> Add the new names with the old config names and essentially duplicate the lines in our sdkconfigs

Although this introduces some (minor) duplication of config IMO it's the simplest thing to do.  The sdkconfig.base already does this for some options, eg with `CONFIG_LOG_BOOTLOADER_LEVEL_WARN`

nevercast, 2019-10-10:
> Although this introduces some (minor) duplication of config IMO it's the simplest thing to do 

It is very simple. The only discomfort I have with this simplist option is that it isn't immune to regression. If names change in the future we have to update them again. If I propose a PR for option 3, use `conf-gen` to handle the rename, would you be interested?

dpgeorge, 2019-10-10:
> If I propose a PR for option 3, use `conf-gen` to handle the rename, would you be interested?

Yes, if it works reliably that would be good.

I think we will want/need to retain support for IDF v3 for some time yet, so would be good to have both v3 and v4 coexist side-by-side cleanly.

nevercast, 2019-10-10:
> would be good to have both v3 and v4 coexist side-by-side cleanly

I'm already having trouble with that as IDF aren't back-porting some fixes.

dmartauz, 2019-10-10:
@nevercast I confirm that changing setting name as recommended above will cause watchdog to not to trigger any more.

jimmo, 2019-10-11:
I've sent a tiny PR to just do the simple rename (like the other existing variables) as this is causing issues with the firmware on micropython.org/download.

I don't think this is particularly burdensome to maintain this approach until 4.x is released, but happy for someone else to do something more sophisticated.

nevercast, 2019-10-11:
Thanks for covering the immediate case. I'll keep the burdensome approach on my backlog for the meantime and probably look at tackling that this weekend.

dpgeorge, 2019-10-11:
Fixed (temporarily) by using duplicate configs in d5cbee3cfbc5fe41e366a55a96e2006fed819d6c

jonnor, 2024-07-27:
There was a fix provided above. And IDF 4.x is no longer used. So I believe this can be closed.

robert-hh, 2024-07-27:
As stated: no support for IDF4.x any more.

PR #5204 (2019-10-11): esp32: Add 4.x version of IDLE WDT config.
Fixes #5200
closed 2024-07-27 by robert-hh