micropython/micropython #5205: NUCLEO_H743ZI build is broken.

iabdalkader, 2019-10-11:
`Traceback (most recent call last):
  File "../../tools/mpy-tool.py", line 834, in <module>
    main()
  File "../../tools/mpy-tool.py", line 822, in main
    raw_codes = [read_mpy(file) for file in args.files]
  File "../../tools/mpy-tool.py", line 822, in <listcomp>
    raw_codes = [read_mpy(file) for file in args.files]
  File "../../tools/mpy-tool.py", line 676, in read_mpy
    raise Exception('incompatible .mpy version')
Exception: incompatible .mpy version
make: *** [../../py/mkrules.mk:120: build-NUCLEO_H743ZI/frozen_mpy.c] Error 1
make: *** Deleting file 'build-NUCLEO_H743ZI/frozen_mpy.c'
make: *** Waiting for unfinished jobs....
`

dpgeorge, 2019-10-12:
mpy-cross needs rebuilding, and also do a `make BOARD=NUCLEO_H743ZI clean` in the stm32 dir.

iabdalkader, 2019-10-12:
That fixed the issue for me, thanks!

closed 2019-10-12 by iabdalkader