micropython/micropython #5207: No rule to make target 'lib/axtls/ssl/asn1.c' while building ESP8266 firmware with Cygwin

iTecAI, 2019-10-12:
Environment: Cygwin on Windows 10

I am attempting to build the micropython esp8266 port with make, but I continue to run into this error:
```
make: *** No rule to make target 'lib/axtls/ssl/asn1.c', needed by 'build-GENERIC/genhdr/qstr.i.last'.  Stop.
```

The full error and debug follows:
```
          Considering target file 'build-GENERIC/genhdr/qstrdefs.collected.h'.
           File 'build-GENERIC/genhdr/qstrdefs.collected.h' does not exist.
            Considering target file 'build-GENERIC/genhdr/qstr.split'.
             File 'build-GENERIC/genhdr/qstr.split' does not exist.
              Considering target file 'build-GENERIC/genhdr/qstr.i.last'.
               File 'build-GENERIC/genhdr/qstr.i.last' does not exist.
                Considering target file 'lib/axtls/ssl/asn1.c'.
                 File 'lib/axtls/ssl/asn1.c' does not exist.
                 Looking for an implicit rule for 'lib/axtls/ssl/asn1.c'.
                 Trying pattern rule with stem 'asn1'.
                 Trying implicit prerequisite 'lib/axtls/ssl/asn1.y'.
                 Trying pattern rule with stem 'asn1'.
                 Trying implicit prerequisite 'lib/axtls/ssl/asn1.l'.
                 Trying pattern rule with stem 'asn1'.
                 Trying implicit prerequisite 'lib/axtls/ssl/asn1.w'.
                 Trying pattern rule with stem 'asn1'.
                 Trying implicit prerequisite 'lib/axtls/ssl/asn1.w'.
                 Trying pattern rule with stem 'asn1.c'.
                 Trying implicit prerequisite 'lib/axtls/ssl/asn1.c,v'.
                 Trying pattern rule with stem 'asn1.c'.
                 Trying implicit prerequisite 'lib/axtls/ssl/RCS/asn1.c,v'.
                 Trying pattern rule with stem 'asn1.c'.
                 Trying implicit prerequisite 'lib/axtls/ssl/RCS/asn1.c'.
                 Trying pattern rule with stem 'asn1.c'.
                 Trying implicit prerequisite 'lib/axtls/ssl/s.asn1.c'.
                 Trying pattern rule with stem 'asn1.c'.
                 Trying implicit prerequisite 'lib/axtls/ssl/SCCS/s.asn1.c'.
                 Trying pattern rule with stem 'asn1'.
                 Trying implicit prerequisite 'lib/axtls/ssl/asn1.y'.
                 Looking for a rule with intermediate file 'lib/axtls/ssl/asn1.y'.
                  Avoiding implicit rule recursion.
                  Trying pattern rule with stem 'asn1.y'.
                  Trying implicit prerequisite 'lib/axtls/ssl/asn1.y,v'.
                  Trying pattern rule with stem 'asn1.y'.
                  Trying implicit prerequisite 'lib/axtls/ssl/RCS/asn1.y,v'.
                  Trying pattern rule with stem 'asn1.y'.
                  Trying implicit prerequisite 'lib/axtls/ssl/RCS/asn1.y'.
                  Trying pattern rule with stem 'asn1.y'.
                  Trying implicit prerequisite 'lib/axtls/ssl/s.asn1.y'.
                  Trying pattern rule with stem 'asn1.y'.
                  Trying implicit prerequisite 'lib/axtls/ssl/SCCS/s.asn1.y'.
                 Trying pattern rule with stem 'asn1'.
                 Trying implicit prerequisite 'lib/axtls/ssl/asn1.l'.
                 Looking for a rule with intermediate file 'lib/axtls/ssl/asn1.l'.
                  Avoiding implicit rule recursion.
                  Trying pattern rule with stem 'asn1.l'.
                  Trying implicit prerequisite 'lib/axtls/ssl/asn1.l,v'.
                  Trying pattern rule with stem 'asn1.l'.
                  Trying implicit prerequisite 'lib/axtls/ssl/RCS/asn1.l,v'.
                  Trying pattern rule with stem 'asn1.l'.
                  Trying implicit prerequisite 'lib/axtls/ssl/RCS/asn1.l'.
                  Trying pattern rule with stem 'asn1.l'.
                  Trying implicit prerequisite 'lib/axtls/ssl/s.asn1.l'.
                  Trying pattern rule with stem 'asn1.l'.
                  Trying implicit prerequisite 'lib/axtls/ssl/SCCS/s.asn1.l'.
                 Trying pattern rule with stem 'asn1'.
                 Trying implicit prerequisite 'lib/axtls/ssl/asn1.w'.
                 Looking for a rule with intermediate file 'lib/axtls/ssl/asn1.w'.
                  Avoiding implicit rule recursion.
                  Trying pattern rule with stem 'asn1.w'.
                  Trying implicit prerequisite 'lib/axtls/ssl/asn1.w,v'.
                  Trying pattern rule with stem 'asn1.w'.
                  Trying implicit prerequisite 'lib/axtls/ssl/RCS/asn1.w,v'.
                  Trying pattern rule with stem 'asn1.w'.
                  Trying implicit prerequisite 'lib/axtls/ssl/RCS/asn1.w'.
                  Trying pattern rule with stem 'asn1.w'.
                  Trying implicit prerequisite 'lib/axtls/ssl/s.asn1.w'.
                  Trying pattern rule with stem 'asn1.w'.
                  Trying implicit prerequisite 'lib/axtls/ssl/SCCS/s.asn1.w'.
                 Trying pattern rule with stem 'asn1'.
                 Rejecting impossible implicit prerequisite 'lib/axtls/ssl/asn1.w'.
                 No implicit rule found for 'lib/axtls/ssl/asn1.c'.
                 Finished prerequisites of target file 'lib/axtls/ssl/asn1.c'.
                Must remake target 'lib/axtls/ssl/asn1.c'.
make: *** No rule to make target 'lib/axtls/ssl/asn1.c', needed by 'build-GENERIC/genhdr/qstr.i.last'.  Stop.
```
given: `make V=1 -d -B`

The normal command (`make V=1`) presents the same error:
```
make: *** No rule to make target 'lib/axtls/ssl/asn1.c', needed by 'build-GENERIC/genhdr/qstr.i.last'.  Stop.
```

My only change to the directory has been to place two python files in the modules/ directory, and I followed the other steps (git update, make mpy-cross) which ran fine.

Is my error just based on the fact that I used Cygwin? Or am I missing some libraries?
Thanks!

dpgeorge, 2019-10-12:
Did you get the submodules?  Try:
```
$ git submodule update --init lib/axtls lib/berkeley-db-1.xx
```

iTecAI, 2019-10-12:
It errors with the message:
```
error: pathspec 'lib/axtls' did not match any file(s) known to git
error: pathspec 'lib/berkeley-db-1.xx' did not match any file(s) known to git
```

I have run `git init` in the directory.

dpgeorge, 2019-10-14:
> I have run `git init` in the directory.

Which directory?  Try to run it in the top-level directory of this repository.

iTecAI, 2019-10-14:
Tried that, got the same error

dpgeorge, 2019-10-14:
I suggest to start from scratch.  git clone, then `git submodule update --init`, then do `ls lib/axtls` to see if there are files there.