micropython/micropython #5209: esp32 missing uos.sync()

poesel, 2019-10-13:
MicroPython v1.11-393-g0096041c9-dirty on 2019-10-04; ESP32 module (spiram) with ESP32
Type "help()" for more information.
> import uos
> uos.sync()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: 'module' object has no attribute 'sync'

Is sync() missing or am I missing something?

robert-hh, 2019-10-13:
The uos module has no sync method. You can see the methods in uos with dir(uos).

poesel, 2019-10-13:
According to this:
https://docs.micropython.org/en/latest/library/uos.html?highlight=uos#module-uos
and this:
https://github.com/micropython/micropython/blob/master/docs/library/uos.rst
it does.

Or is the documentation wrong?

robert-hh, 2019-10-13:
So it is a litlle bit more complicated. Looking in to the devices, I found, that
- the STM32 port has uos.sync()
- the esp32 and esp8266 do not have uos.sync()
- the CC3200 port (WiPy1) has uos.sync()
Since a while the documentation is pretty misleading, since it does not tell which feature is supported by which board. There is just a disclaimer that not every feature is supported by every port. So you have to try.

poesel, 2019-10-13:
Ah, ok. Thanks for the information.

closed 2019-10-13 by poesel