micropython/micropython #5213: ESP32 freezing if there is no slave SPI device connected

Qudor-Engineer, 2019-10-15:
Hello everybody,
When I use SPI (Hardware or software) as a master in ESP32 and there is no slave device connected to the SPI of the ESP32 the code will stuck on "SPI readinto" function for infinity even when the timeout was set. I used all available exceptions to get the error but actually the exception didn't rise. I used different types of ESP32 but still the same problem. Also if the code was working and suddenly the slave SPI device was removed or turned off the ESP32 freezing on "SPI readinto" function forever. I used micropython firmware versions "esp32-20190529-v1.11" and "esp32-20191011-v1.11-422-g98c2eabaf (latest)"

MrSurly, 2019-10-15:
@Qudor-Engineer 

Can you paste code to reproduce, please?

Qudor-Engineer, 2019-10-16:
Hi
@MrSurly 
thank you for the fast replay
this is my code to read pressure value from SPI pressure sensor 

from machine import Pin, SPI
from time import sleep
import esp
import struct

def pressure_sensor_init():
  sleep(0.5)

  # Defining sensor pins (EOC & chip select)
  global sensor_eoc, sensor_cs, spi
  sensor_eoc = Pin(4, Pin.IN)
  sensor_cs = Pin(5, Pin.OUT)
  sensor_rst = Pin(2, Pin.OUT)
  
  sensor_rst.on()
  sleep(0.05)
  sensor_rst.off()
  sleep(0.15)
  sensor_rst.on()
  sensor_cs.on()

  # Initialize the Hardware SPI 2
  spi = SPI(2, baudrate=400000, polarity=0, phase=0, bits=8, firstbit=SPI.MSB, sck=Pin(18), mosi=Pin(23), miso=Pin(19)) 
  
#==============================================================================================================#

# Request presure value from the sensor (in PSI)
def request_pressure_value():
  
  # chip select
  sensor_cs.off()
  
  # write these addresses to the pressure sensor to make it starts masurments
  addresses = bytearray((0xAA, 0x00, 0x00))
  spi.write(addresses)
  
  # chip deselect
  sensor_cs.on()
  
# Retrive presure value from the sensor (in PSI)
def retrive_pressure_value():
    
  # wait the senser until complete the masurments (EOC pin will be "HIGH" when the senser complete the masurments)
  while not sensor_eoc.value():
    pass
  
  addresses2 = bytearray((0xF0, 0x00, 0x00, 0x00))
  received_data = bytearray(3)
  
  # chip select
  sensor_cs.off()
  
  # Retrive the data from the senser (masured pressure)
  buf, buf1, buf2, buf3 = bytearray(1), bytearray(1), bytearray(1), bytearray(1)
  spi.readinto(buf , addresses2[0])    # it stuck on this line forever if the sensor removed or turned off
  spi.readinto(buf1, addresses2[1])
  spi.readinto(buf2, addresses2[2])
  spi.readinto(buf3, addresses2[3])
  
  # concrete the retrived data into one bytearray
  received_data = buf1 + buf2 + buf3
  
  # chip deselect
  sensor_cs.on()
  
  # Convert the retrived data from bytes to integer
  raw_pressure = int.from_bytes(received_data, 'big')
  
  # calculate the pressure in PSI using manufacturer formula (from the datasheet)
  psi_pressure = ((raw_pressure - 1677722) * 15) / (15099494-1677722)
  
  return psi_pressure
  
  
#==============================================================================================================#
  
  
if __name__ == "__main__":
  # initializing the senser  
  pressure_sensor_init()
  
  # request pressure value
  request_pressure_value()

  # get pressure value from the senser
  print("pressure (psi) = {}".format(retrive_pressure_value()))

MrSurly, 2019-10-16:
@Qudor-Engineer 

If you power up your ESP32 without the sensor attached, does it do this?

Or does it only do this if the sensor is attached, then removed?

Qudor-Engineer, 2019-10-17:
@MrSurly 
> 
> If you power up your ESP32 without the sensor attached, does it do this?
> 
> Or does it only do this if the sensor is attached, then removed?

Yes, when I power up my ESP32 without attaching the sensor it does that.
I tried all probabilities and got the same results "it stuck forever".

dpgeorge, 2019-10-22:
It's very strange for the code to freeze on a SPI read, because it doesn't know/care if a slave is connected or not.

I tried the above code on a Lolin32 Pro board with `MicroPython v1.11-487-g9c5262f25`, build with IDF v3.3.  Without anything attached it runs to completion and prints out `pressure (psi) = -1.875001`

Qudor-Engineer, 2019-10-23:
Hello @dpgeorge 
I'm using ESP-WROOM-32 three boards and 
ESP-WROOM-32D ( ESP-PoE from Olimex) two boards and I flashed the latest firmware (esp32-20191023-v1.11-497-gf301170c7.bin) and I still face the same problem

Qudor-Engineer, 2019-10-28:
Hello @dpgeorge and @MrSurly 
when I tried the code with Heltec ([WiFi LoRa](https://heltec.org/project/wifi-lora-32/)) that use ESP3212 it work and not stuck but with other models of ESP32 that I mentioned them above the code stuck

dpgeorge, 2019-10-29:
This really sounds like a hardware problem, with "other models of ESP32".  The way forward would be to try out a simple ESP IDF example that uses SPI and see if the issue is still there.  If so then it's a problem for Espressif, otherwise a problem for this repo.