micropython/micropython #5226: BLE: crash after running gc.collect()

dpgeorge, 2019-10-18:
Issue split off from https://github.com/micropython/micropython/pull/5171#issuecomment-543075970

----

I did find however, that the garbage collection is deleting something important to do with BLE. This bug can be replicated by using the snippet sample above by @HexVitor [see below] and using the following steps:

1. Establish a BLE connection using something like Serial Bluetooth Terminal on Android
1. Send a couple of test messages successfully from Serial Bluetooth Terminal
1. Type gc.collect() at the REPL
1. Try to send a few more messages and you should eventually see something like the following...

```
Guru Meditation Error: Core  0 panic'ed (InstrFetchProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x3ffe67b0  PS      : 0x00060530  A0      : 0x80148fdc  A1      : 0x3ffbb4a0  
A2      : 0x3ffc5cbc  A3      : 0x00000010  A4      : 0x00000000  A5      : 0x3ffbb4e0  
A6      : 0x3ffc80ec  A7      : 0x3ffe67b0  A8      : 0x80148fa1  A9      : 0x00000000  
A10     : 0x00000000  A11     : 0x00000010  A12     : 0x3ffbb4e0  A13     : 0x00000002  
A14     : 0x0000cdcd  A15     : 0x00000001  SAR     : 0x00000018  EXCCAUSE: 0x00000014  
EXCVADDR: 0x3ffe67b0  LBEG    : 0x4014d139  LEND    : 0x4014d144  LCOUNT  : 0x00000009  

ELF file SHA256: 0000000000000000000000000000000000000000000000000000000000000000

Backtrace: 0x3ffe67ad:0x3ffbb4a0 |<-CORRUPTED
```

Code:
```python
# Import of modules and classes
import bluetooth

# Configuration
ble = bluetooth.BLE()
ble.active(True)

# Event Handling
def ble_irq(event, data):
    # print received data
    print(ble.gatts_read(rx))
        
ble.irq(ble_irq)

# GATT Server
UART_UUID = bluetooth.UUID('6E400001-B5A3-F393-E0A9-E50E24DCCA9E')
UART_TX = (bluetooth.UUID('6E400003-B5A3-F393-E0A9-E50E24DCCA9E'), bluetooth.FLAG_READ | bluetooth.FLAG_NOTIFY,)
UART_RX = (bluetooth.UUID('6E400002-B5A3-F393-E0A9-E50E24DCCA9E'), bluetooth.FLAG_WRITE,)
UART_SERVICE = (UART_UUID, (UART_TX, UART_RX,),)
SERVICES = (UART_SERVICE,)
((tx, rx,), ) = ble.gatts_register_services(SERVICES)

# Advertiser
def adv_encode_name(name):
    name = bytes(name, 'ascii')
    return bytearray((len(name) + 1, 0x09)) + name

ble.gap_advertise(100, bytearray('\x02\x01\x02') + adv_encode_name('ESP32'))
```

jimmo, 2019-10-21:
Thanks for the report @sunbeamer !

I can reproduce this on ESP32 but (so far) not on PYBD. Will investigate and get back to you.

sunbeamer, 2019-10-22:
Thanks for the quick fix @jimmo !

I have tested your fix, both using the steps to reproduce the bug I outlined above, as well as in my own application code that first highlighted the bug and all works OK.

commit f34e16db (2019-10-22): extmod/modbluetooth: Persist reference to NimBLE service instances.

NimBLE doesn't actually copy this data, it requires it to stay live.
Only dereference when we register a new set of services.

Fixes #5226

This will allow incrementally adding services in the future, so
rename `reset` to `append` to make it clearer.
PR #5240 (2019-10-22): extmod/modbluetooth: Persist reference to NimBLE service instances.
NimBLE doesn't actually copy this data, it requires it to stay live.
Only dereference when we register a new set of services.

Fixes #5226

This will allow incrementally adding services in the future, so
rename `reset` to `append` to make it clearer.

Also fix a truncation with chars/descs that have been increased
from the default size.
closed 2019-10-22 by dpgeorge