micropython/micropython #5231: OSError: Error running mpy-cross, rc=1

LukasDrsman, 2019-10-19:
Error running mpy-cross when using micropython-py_compile

dpgeorge, 2019-10-22:
Please provide more information about the problem including steps to reproduce it.

fragmuffin, 2020-05-05:
I can't speak for @LukasDrsman, but I experienced a similar problem with:

```python
import os
from fnmatch import filter
import mpy_cross

for pyfile in filter(os.listdir('.'), '*.py'):
    mpy_cross.run(pyfile)
```

This failed on roughly every 2nd iteration presumably because the _process_ returned by `mpy_cross.run` was still running when the next iteration came around.

**Solution**: add `.wait()` to theh `mpy_cross.run()` call:

```python
mpy_cross.run(pyfile).wait()  # <--
```

arlucio, 2020-09-10:
Just made this to turn all .py files in a directory to .mpy and than save them all to device using mpfshell. Works perfectly!

```
import os
import mpy_cross

dir = "[YOUR_DIRECTORY]"
port = "COM4"
cmd_str = 'open %s;' % port

for file in os.listdir(dir):
    if file.endswith(".py"):
        mpy_cross.run(dir+file)
        cmd_str = cmd_str + 'put %s;' % file

os.system('mpfshell -n -c %s' % cmd_str)
```

dpgeorge, 2020-09-11:
> Just made this to turn all .py files in a directory to .mpy

See also `tools/mpy_cross_all.py`

Ardutec, 2021-01-24:
I am also facing a somewhat similar issue. 

I need to hide my micropython file containing some functions and need to access the functions from another file. For this, I need to convert my micropython code into .pyc file and I was using py_compile library for this, but it is giving me this error.

Micropython Code:
import py_compile
py_compile.compile(file.py)

Error:
sh: 1: mpy-cross: not found
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/pi/.micropython/lib/py_compile.py", line 12, in compile
OSError: Error running mpy-cross, rc=12

I am using micrpython unix port on Raspberry Pi.
micropython version v1.13.243-g032e09562-dirty
mpy-cross mpy v5

robert-hh, 2021-01-24:
It's nto clear on which device you want to run mpy-cross. mpy-cross is intended to run on your host computer (linux, windows or mac). You can start it from the bash command line.

Ardutec, 2021-01-25:
I am not running mpy-cross directly. I am using a micropython library "py_compile" for creating .pyc file. Running the micropython from the command line with command /home/pi/micropython/ports/unix/micropython,  on Raspberry Pi 3 Model B, Rasbian baster.

peterhinch, 2021-01-25:
> I need to convert my micropython code into .pyc file

`py_compile` is a CPython library and `.pyc` are CPython precompiled files. The MicroPython equivalents are `mpy-cross` and `.mpy`. They are entirely different.

robert-hh, 2021-01-25:
Since you get this "mpy-cross: not found" error, it looks like the process is trying to call mpy-cross. Do you have the mpy-cross executable in one of the directories listed in PATH? Did you build mpy-cross?

Ardutec, 2021-01-25:
> py_compile is a CPython library and .pyc are CPython precompiled files. The MicroPython equivalents are mpy-cross and .mpy. They are entirely different.

Thank-you, yes they are different. Python3 py_compile library was exporting a .pyc file but micropython py_compile library was not working as @robert-hh suggested due to path issue, mpy-cross was buil but not added to PATH.
Now I have converted .py file into .mpy file with the given command and it is fulfilling my purpose.
`cd ~`
`cd ./micropython/mpy-cross`
`./mpy-cross -march=armv6 -mcache-lookup-bc ./file.py # make .mpy file and by putting in same directory`

jonnor, 2024-09-19:
Not enough information to reproduce original problem. And the later discussed problem looks to have been resolved.