micropython/micropython #5232: BLE/ESP32: limit to number of services?

poesel, 2019-10-19:
It seems to me that the number of advertised services is limited to three.
This:
`        self._payload = advertising_payload(name=name, services=([0x1800,0x180F,0x181A,]), appearance=_ADV_APPEARANCE_GENERIC_WEIGHT_SCALE)
`
works. While this:
        `self._payload = advertising_payload(name=name, services=([0x1800,0x180F,0x181A,0x181D,]), appearance=_ADV_APPEARANCE_GENERIC_WEIGHT_SCALE)
`
won't.

Fault message is:
`  File "main.py", line 100, in _advertise
OSError: [Errno 5] EIO
MicroPython v1.11-427-gd5cbee3cf-dirty on 2019-10-14; ESP32 module (spiram) with ESP32`

I found nothing about a limit in the documentation. 
Is this a hardware limit or something that can be configured?

dpgeorge, 2019-10-21:
The GAP advertising payload has a limit of 37 bytes (for BLE=4.x) so, depending on the length of `name`, it's probably just over that limit with more services.

Cubic-Rambo, 2019-10-21:
nRF52832 Boards How can it support bluetooth nimble? @dpgeorge

jimmo, 2019-10-21:
Yeah, the payload length the problem here.

I believe the solution here is to use a combination of the `BLE_GAP_AD_TYPE_16BIT_SERVICE_UUID_MORE_AVAILABLE` (0x02) type, and/or using active scan (via the scan_resp param) to allow more advertising data. I plan to write up an example using this.

poesel, 2019-10-21:
Thanks, that was the problem. The name was 8 characters long. Changing this to 1 allowed 4 services to be advertised.

closed 2019-10-21 by poesel

Walkline80, 2020-03-02:
> Thanks, that was the problem. The name was 8 characters long. Changing this to 1 allowed 4 services to be advertised.

You can also passing "name" param by resp_data
[doc link](http://docs.micropython.org/en/latest/library/ubluetooth.html?highlight=bluetooth#ubluetooth.BLE.gap_advertise)