micropython/micropython #5233: (coroutine of object) != (courotine of object), CPython incompatibility

kevinkk525, 2019-10-20:
Sorry for not being able to be more specific in the issue description, suggestions welcome.

The coroutine of an object changes its identity every time it gets accessed:
```
class C:
    async def test(self):
        pass

>>> c=C()
>>> c.test==c.test
False
>>> a=c.test
>>> b=c.test
>>> a==b
False
```

Therefore there is no way of storing a link to the coroutine and later comparing if the stored link actually belongs to the courotine of that class.

In Cpython this of course works:
```
>>> class C:
...     async def test(self):
...        pass
>>> c=C()
>>> a=c.test
>>> b=c.test
>>> a
<bound method C.test of <__main__.C object at 0x0000019BCEA08E80>>
>>> b
<bound method C.test of <__main__.C object at 0x0000019BCEA08E80>>
>>> a==b
True
>>> c.test==c.test
True
>>> 
```

When comparing the coroutines of a class then it works correctly:
```
>>> C.test==C.test
True
>>> a=C.test
>>> b=C.test
>>> a==b
True
```

With coroutines outside of classes it does work too:
```
async def test():
...     pass
>>> a=test
>>> b=test
>>> a==b
True
>>> test==test
```

Is this behaviour fixable or a victim of optimizations and resource savings?

jimmo, 2019-10-21:
Unless I'm missing something, I think this is true of non-coroutine methods too.

```
>>> class B:
...     def foo(self):
...         pass
...         
...         
... 
>>> b = B()
>>> b.foo == b.foo
False
>>> B.foo == B.foo
True
>>> 
```

dpgeorge, 2019-10-21:
The inequality is because bound-method objects are created on the heap each time they are needed, and so are "different" because they are different entities.

I'm not sure how CPython makes them the same... it could be caching the bound-method objects so creating it a second time returns the same instance as the first one, or it could be implementing the equality operator on these.

seandepagnier, 2020-05-30:
Perhaps bound-method objects can just create a pointer to the object plus the method on the heap if they are assigned and not called.   In this way comparisons could work without using much more memory.

dpgeorge, 2020-06-02:
> I'm not sure how CPython makes them the same... it could be caching the bound-method objects so creating it a second time returns the same instance as the first one, or it could be implementing the equality operator on these.

It's implementing the equality operator on bound method objects, which checks that both the self and the method are the same.  Probably that's how uPy would need to do it if this feature is important/needed.

> Perhaps bound-method objects can just create a pointer to the object plus the method on the heap if they are assigned and not called.

Yes that's how they are currently implemented; see `py/objboundmeth.c`.

BrianPugh, 2023-02-27:
Here's another example with just normal methods:

```
class MyClass:
    def __init__(self):
        self.ref1 = self.method
        self.ref2 = self.method

    def method(self):
        pass

my_class = MyClass()

print(my_class.ref1 == my_class.ref2)
```

Outputs:
```
$ python mp_test.py
True
$ micropython mp_test.py
False
```

BrianPugh, 2023-02-27:
It's not 100% logically correct, but I could mitigate it in my calling code for now by checking:

```
my_class.ref1.__name__ == my_class.ref2.__name__
```

dpgeorge, 2023-10-13:
Fixed by 4f5e165d0bf9ba338df3cdab4266556c4c5a70c7

closed 2023-10-13 by dpgeorge