micropython/micropython #5238: nrf build failure - 12413e

diginfo, 2019-10-21:
Just downloaded and clean built the latest (12413e92a3f938bdfe844bd65bf189ee1431e1d0) for esp32, esp8266, pyboard, sf2 and sf6, however it fails when building nrf as follows:

These updates and builds are all done using bash scripts, I was not able to determine from the errors if this is anything to do with my setup ??

```
@@ Building micropython ...
make: Entering directory '/nfs/qnap/dev/uPython/build/micropython/micropython/mpy-cross'
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
make: Leaving directory '/nfs/qnap/dev/uPython/build/micropython/micropython/mpy-cross'
make clean...
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
rm -rf build-pca10040 
/bin/bash: warning: setlocale: LC_ALL: cannot change locale (uk_UA.UTF-8)
make BOARD=pca10056 SD=s140 FROZEN_MPY_DIR=freeze MICROPY_FATFS=1
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
GEN build-pca10056-s140/frozen_mpy.c
Traceback (most recent call last):
  File "../../tools/mpy-tool.py", line 833, in <module>
    main()
  File "../../tools/mpy-tool.py", line 821, in main
    raw_codes = [read_mpy(file) for file in args.files]
  File "../../tools/mpy-tool.py", line 821, in <listcomp>
    raw_codes = [read_mpy(file) for file in args.files]
  File "../../tools/mpy-tool.py", line 688, in read_mpy
    return read_raw_code(f, qstr_win)
  File "../../tools/mpy-tool.py", line 621, in read_raw_code
    prelude = read_prelude(f, fun_data, qstr_win)
  File "../../tools/mpy-tool.py", line 589, in read_prelude
    read_qstr_and_pack(f, bytecode, qstr_win) # simple_name
  File "../../tools/mpy-tool.py", line 596, in read_qstr_and_pack
    qst = read_qstr(f, qstr_win)
  File "../../tools/mpy-tool.py", line 560, in read_qstr
    return qstr_win.access(ln >> 1)
  File "../../tools/mpy-tool.py", line 86, in access
    val = self.window[idx]
IndexError: list index out of range
make: *** [../../py/mkrules.mk:126: build-pca10056-s140/frozen_mpy.c] Error 1
make: *** Deleting file 'build-pca10056-s140/frozen_mpy.c'
```

jimmo, 2019-10-21:
I normally see this when I try and run mpy-tool on a .mpy that was built with an incorrect version of mpy-cross.

Looking at that build output, it looks like the `make clean` (which outputs `rm -rf build-pca10040`) is cleaning a different board config than the one that you build. `make BOARD=pca10056 ...`.

dpgeorge, 2019-10-22:
Yes, a `make clean` and `make -C mpy-cross` should be enough to get this working.

closed 2019-10-22 by dpgeorge