micropython/micropython #5239: Build for Pyboard V1.x does not handle frozen modules

peterhinch, 2019-10-21:
This is since the manifest mod. Modules in the stm32/modules directory are cross compiled but they aren't included in the build. For example the build output includes
```
MPY modules/font10.py
```
but I can't import it and it doesn't appear in help('modules'). 
I notice that at the end of the output there is
```
MPY dht.py
MPY lcd160cr.py
MPY lcd160cr_test.py
MPY onewire.py
GEN build-PYBV10/frozen_content.c
CC build-PYBV10/frozen_content.c
```
These modules do appear as frozen modules. I suspect that my attempt to override the manifest file works only for Pyboard D.

My build script works on Pyboard D:
```bash
MPDEVICE='/dev/pyboard'

# Determine board type
BOARD=$(pyb_check $MPDEVICE)

# Check for user override of frozen directory
if [ $FROZEN_DIR ]
then
    echo Frozen modules located in $FROZEN_DIR
else
    FROZEN_DIR='modules'
fi

if [ $BOARD ]
then
    echo Building for $BOARD
    cd $MPDIR/ports/stm32
    if [ $# -eq 1 ] && [ $1 = "--clean" ]
    then
        make BOARD=$BOARD clean
    fi
    if make -j 8 BOARD=$BOARD FROZEN_MPY_DIR=$FROZEN_DIR FROZEN_MANIFEST= && pyb_boot $MPDEVICE
    then
        sleep 1
        make PYTHON=python3 BOARD=$BOARD deploy
    else
        echo Build failure
    fi
    cd -
else
    echo Incorrect board type
fi
```
Full listing:
```
Building for PYBV10
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
mkdir -p build-PYBV10/genhdr
make: Warning: File 'build-PYBV10/' has modification time 13 s in the future
mkdir -p build-PYBV10/build-PYBV10/
mkdir -p build-PYBV10/drivers/bus/
mkdir -p build-PYBV10/drivers/dht/
mkdir -p build-PYBV10/drivers/memory/
mkdir -p build-PYBV10/extmod/
mkdir -p build-PYBV10/lib/embed/
mkdir -p build-PYBV10/lib/libc/
mkdir -p build-PYBV10/lib/libm/
mkdir -p build-PYBV10/lib/mp-readline/
mkdir -p build-PYBV10/lib/netutils/
mkdir -p build-PYBV10/lib/oofatfs/
mkdir -p build-PYBV10/lib/stm32lib/CMSIS/STM32F4xx/Source/Templates/
mkdir -p build-PYBV10/lib/stm32lib/CMSIS/STM32F4xx/Source/Templates/gcc/
[... 347 lines cut ...]
MPY dht.py
MPY lcd160cr.py
MPY lcd160cr_test.py
MPY onewire.py
GEN build-PYBV10/frozen_content.c
CC build-PYBV10/frozen_content.c
LINK build-PYBV10/firmware.elf
   text	   data	    bss	    dec	    hex	filename
 335188	     32	  27844	 363064	  58a38	build-PYBV10/firmware.elf
GEN build-PYBV10/firmware.dfu
Writing build-PYBV10/firmware.dfu to the board
File: build-PYBV10/firmware.dfu
    b'DfuSe' v1, image size: 335525, targets: 1
    b'Target' 0, alt setting: 0, name: "ST...", size: 335240, elements: 2
      0, address: 0x08000000, size: 14872
      1, address: 0x08020000, size: 320352
    usb: 0483:df11, device: 0x0000, dfu: 0x011a, b'UFD', 16, 0x9377312e
Writing memory...
0x08000000   14872 [=========================] 100% 
0x08020000  320352 [=========================] 100% 
Exiting DFU...
make: warning: Clock skew detected. Your build may be incomplete.
/mnt/qnap2/data/Projects/MicroPython/micropython_ra8875
[adminpete@capybara]: /mnt/qnap2/data/Projects/MicroPython/micropython_ra8875
```

jimmo, 2019-10-21:
I think the issue here is that the `make` is using your custom modules directory, but then the `make deploy` is using the default manifest. The `deploy` step re-links the elf file, with frozen_content.c (which comes from a manifest), whereas the main build links with frozen_mpy.c (which comes from FROZEN_MPY_DIR).

Can you try adding `FROZEN_MPY_DIR=$FROZEN_DIR FROZEN_MANIFEST=` to the `make deploy` line too.

peterhinch, 2019-10-21:
That works on Pyboard 1.x and Pyboard D. Thank you.

closed 2019-10-21 by peterhinch