micropython/micropython #5242: Cancelling coroutines: can't pend throw to just-started generator.

kevinkk525, 2019-10-21:
My code heavily relies on coroutines which get canceled quite often.
Occasionally it happens, that a coroutine gets put on the loop but then canceled before it gets executed for the first time.
Apparently this is not possible as it throws a TypeError:
```
import uasyncio as asyncio
>>> async def test():
...     print("test")
>>> t=test()
>>> async def cancel():
...     asyncio.cancel(t)
>>> loop=asyncio.get_event_loop()
>>> loop.create_task(cancel())
>>> loop.create_task(t)
>>> loop.run_forever()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "uasyncio/core.py", line 109, in run_forever
  File "<stdin>", line 2, in cancel
  File "uasyncio/core.py", line 273, in cancel
TypeError: can't pend throw to just-started generator
```

This is incompatible to CPython and causes some problems for me because suddenly there is no way to cancel this coroutine unless I wait until it yields for the first time. 
An example of where this might happen is if an asyncio task is created when an MQTT connection is established and the same task gets cancelled when the connection breaks. So if the connection breaks until the task gets executed, it would run into this error.

Is this fixable or just the way generators work in micropython?

dpgeorge, 2019-10-22:
> Is this fixable or just the way generators work in micropython?

MicroPython uses `pend_throw()` on generators (an extension to CPython) to raise an exception into a generator the next time it is iterated/yielded upon.  This will only work for generators that have started.

Note that a generator can catch the injected `CancelledError` and ignore it, ie not get cancelled, or delay the cancellation.  That's only possible because the generator has started execution.

The way cancellation of tasks is exposed to the user in `uasyncio` is different to how CPython does it, and the behaviour seen above is a consequence of this difference.

Do you know in CPython if it's possible to do this, cancel a task before it has started?  Does it inject an exception or just never even start the generator/task?

kevinkk525, 2019-10-22:
Since a generator can only be cancelled after it has first yielded, it could have executed a lot of synchronous code before the cancellation even gets to it. So as half-baked workaround every coroutine would need to yield right at the start to ensure it can catch a Cancellation right at the start while the caller (the coroutine that wants to cancel the generator) has to try to cancel it until it doesn't throw an error. So a workaround in micropython would look like this:
```
async def foo():
    await asyncio.sleep(0)  # yield once to make it cancellable
    try:
        print("foo")
        await asyncio.sleep(1)
    except asyncio.CancelledError:
        print("foo got canceled")

async def main():
    t = foo()
    loop.create_task(t)
    while True:
        try:
            asyncio.cancel(t)
            break
        except TypeError:  # can't pend throw to just-started generator
            await asyncio.sleep(0)
    await asyncio.sleep(1)
    print("main done")

loop.run_until_complete(main())
```

I know CPython does it differently with Tasks but they can be canceled before they even yield and then never get executed at all (so it also doesn't inject the CancelledError, the Task just doesn't get executed)
```
import asyncio

async def foo():
    try:
        print("Foo started")
        await asyncio.sleep(1)
        print("Foo done")
    except asyncio.CancelledError:
        print("Foo got canceled")

async def main():
    print("main started")
    task = asyncio.ensure_future(foo())
    task.cancel()
    await asyncio.sleep(2)
    print("main done")

loop = asyncio.get_event_loop()
loop.run_until_complete(main())

>>> main started
>>> main done
```

dxxb, 2019-10-22:
> So as half-baked workaround every coroutine would need to yield right at the start to ensure it can catch a Cancellation right at the start while the caller

Would this also work for you? My thinking is: if the generator hasn't started there is nothing to cancel.

``` py
def my_task_cancel(t):
    try:
        asyncio.cancel(t)
    except TypeError:
        pass
```

I wish `pend_throw()` raised a more specific exception in case the generator had not started yet.

kevinkk525, 2019-10-22:
No that doesn't work. 

> Would this also work for you? My thinking is: if the generator hasn't started there is nothing to cancel.

The logic here is wrong. A generator that hasn't been started is already in the uasyncio loop and will therefore be started as soon as the loop gets to it. 
As soon as you call
```
loop.create_task(t)
```
the generator is out of your control and will get started.

dxxb, 2019-10-22:
> the generator is out of your control and will get started.

You are correct, I was confused about the semantics of `pend_throw()`. I implemented task cancellation in an old version of uasyncio (years ago!) when `pend_throw()` didn't exist.

Next attempt:

``` py

import uasyncio as asyncio

class MyExc(Exception):
    pass

def my_task_cancel(t):
    try:
        asyncio.cancel(t)
    except TypeError as e:
        try:
            print("Caught <{}>".format(e))
            t.throw(MyExc("Canceling {} synchronously".format(t)))
        except MyExc as e:
            print(e)
    else:
        print("Canceling {} asynchronously".format(t))

async def test1():
    try:
        print("test1.1")
        await asyncio.sleep(1)
        print("test1.2")
        await asyncio.sleep(1)
    except Exception as e:
        print("test1", type(e), e)
        raise

async def test2():
    try:
        print("test2.1")
        await asyncio.sleep(1)
        print("test2.2")
        await asyncio.sleep(1)
    except Exception as e:
        print("test2", e)
        raise

async def dump():
    while True:
        loop = asyncio.get_event_loop()
        print("===== Loop at", loop.time(), "ms, r:", len(loop.runq), "w:", len(loop.waitq))
        if len(loop.runq) == 0 and len(loop.waitq) == 0:
            break
        await asyncio.sleep(0.5)

d = dump()
t1 = test1()
t2 = test2()

async def cancel():
    my_task_cancel(t1)
    my_task_cancel(t2)

loop=asyncio.get_event_loop()
loop.create_task(t1)
loop.create_task(cancel())
loop.create_task(t2)
loop.run_until_complete(d)
```

Which seems to work in the surface...

``` sh
$ ./micropython task_cancel_test.py
test1.1
Canceling <generator object 'test1' at b7c74a70> asynchronously
Caught <can't pend throw to just-started generator>
Canceling <generator object 'test2' at b7c74e40> synchronously
===== Loop at 18976639 ms, r: 0 w: 1
===== Loop at 18977141 ms, r: 0 w: 1
test1 <class 'CancelledError'>
===== Loop at 18977645 ms, r: 0 w: 0
```

However, how does current uasyncio deal with task cancelation when a task is registered with a poller? I recall that dealing with that case required some changes to the event loop core and it wasn't pretty.

kevinkk525, 2019-10-22:
@dxxb This is indeed a workaround that will work too. Thanks for that.
Using this for task cancellation could be helpful in lack of a better way:
```
def cancel(t):
   try:
        asyncio.cancel(t)
    except TypeError as e:
        try:
            print("Caught <{}>".format(e))
            t.throw(MyExc("Canceling {} synchronously".format(t)))
        except MyExc as e:
            print(e)
    else:
        print("Canceling {} asynchronously".format(t))
```

If generators can't receive a pend_throw before they are started, then what @peterhinch suggests seems to be the cleaner way, finally introducing the fixes to runq and waitq as per Paul's fixes.
That way we'd now get rid of 2 bugs in uasyncio.
The set is a nice workaround but one that would have to be done on uasyncio module level so when messing around there, you could just as well fix timeq module for waitq and deque for runq.

peterhinch, 2019-10-22:
I have prototyped this approach in my fast_io fork with some success. I adapted it from my suggestion above so it works with official firmware.

The **EventLoop** class has a **.pending** set: task ID's are added to this before the task is placed on any queue. **EventLoop** already has a **.canned** set of tasks to be cancelled. **cancel()** first checks if the task is in **.pending**: if so it removes it from **.pending**, adds it to **.canned** and quits without running **pend_throw**.

In the **EventLoop.run_forever()** loop, before a task is moved from any of the queues to runq it is checked to see if it is in **.canned**: if so it is removed from **.canned** and the task is thrown away. This is the existing cancellation method. Now, when a task is popped from runq it is checked for membership of **.canned**: if so it is thrown away. This caters for the case of a task which has been put on runq directly (by **yield** or **sleep(0)**) and which hasn't yet run.

This has passed simple tests and has run full applications but it needs more testing. It has the drawback of excessive allocation events: **.canned** was used only when a task is cancelled whereas **.pending** is used every time a task is scheduled, even if the application does no cancellation. So a coro like
```python
async def foo():
    while True:
        await asyncio.sleep(0)
```
will be constantly performing allocations and de-allocations. This is very unfortunate and applies equally to my suggestion above using Paul's better cancellation method.

I appreciate you want to use official uasyncio but it shows a way towards a fix. But I do hope someone thinks of a better way...

kevinkk525, 2019-10-23:
Your implementation has some "nice" workarounds but the cancel will still fail on a just-started generator as your cancel starts with:
```
def cancel(coro):
    prev = coro.pend_throw(CancelledError())
```
But after that the logic seems right. 

It's however quite the workaround for a problem that should be fixed in the underlying libraries. 
I hope we don't have to wait years for a fix. I'd create PRs for utimeq and deque but I'm afraid that all I could contribute would be copying pfalcons implementation.. So better someone else looks at it.
We need a better cancellation implementation for uasyncio.

jimmo, 2019-10-29:
Thanks for raising this @kevinkk525 and the suggestions from @peterhinch and @dxxb .

I think the best approach here is to just fix pend_throw to work for unstarted coroutines. This way we avoid having to change uasyncio or risk breaking anything or do any additional allocations. I think I see a way to implement this efficiently in objgenerator.c.

kevinkk525, 2019-10-29:
That would be great. Thanks.
Basically throwing an exception to the generator directly like @dxxb suggested works exactly in the way one would expect it to work, the generator goes straight to a StopIteration when uasyncio tries to execute it and therefore no code of the generator gets executed.

However the cancellation in uasyncio still has issues (only cancelling after a long yield has finished) and would need work too although sadly it seems that there's little interest in changing uasyncio and the underlying modules.

jimmo, 2019-10-29:
See #5275

> However the cancellation in uasyncio still has issues (only cancelling after a long yield has finished) and would need work too although sadly it seems that there's little interest in changing uasyncio and the underlying modules.

I agree there are lots of things that can be improved, I've got a few on my todo list, but can you provide more information about this specific issue? There are other changes in the works for uasyncio (e.g. #5222).

kevinkk525, 2019-10-29:
Thanks for the fix.

> I've got a few on my todo list, but can you provide more information about this specific issue?
See #5276

peterhinch, 2019-10-29:
#5276 is the most serious issue. But also see these major bugs:
[Long I/O latency](https://github.com/micropython/micropython-lib/pull/287)
[Bug with full duplex I/O](https://github.com/micropython/micropython/pull/3836#issuecomment-397317408)
For completeness there is the [modlwip issue](https://github.com/micropython/micropython/pull/5222).

And some simple but useful enhancements:
[run_until_complete returns value as per CPython](https://github.com/micropython/micropython-lib/pull/270)
[Fails silently if generator function is launched](https://github.com/micropython/micropython-lib/pull/292)

[The start of this doc](https://github.com/peterhinch/micropython-async/blob/master/FASTPOLL.md) lists some other minor enhancements.

dxxb, 2019-10-29:
> However the cancellation in uasyncio still has issues (only cancelling after a long yield has finished) and would need work too although sadly it seems that there's little interest in changing uasyncio and the underlying modules.

@kevinkk525, I am interested in fixing this but had not looked at the status of uasyncio in a long time.

> I'd create PRs for utimeq and deque but I'm afraid that all I could contribute would be copying pfalcons implementation.

I will try to make some time in the following days to create a PR for this.

> I think the best approach here is to just fix pend_throw to work for unstarted coroutines. This way we avoid having to change uasyncio or risk breaking anything or do any additional allocations.

@jimmo, thank you for the great PR including the additional tests. The upsides of your changes seem compelling to me but I wonder if there is any downside in preventing the task from handling the exception injected by `pend_throw()` in case it hasn't started yet. Does anyone recall why the existing implementation of `pend_throw()` is limited to injecting exceptions when the generator has started?

@peterhinch, thank you for providing a summary of outstanding issues I was looking for that!

kevinkk525, 2019-10-29:
> @jimmo, thank you for the great PR including the additional tests. The upsides of your changes seem compelling to me but I wonder if there is any downside in preventing the task from handling the [...]

CPython prevents the task from handling the exception if the coroutine hasn't started yet so this is the expected behaviour.

dxxb, 2019-10-29:
> CPython prevents the task from handling the exception if the coroutine hasn't started yet so this is the expected behaviour.

https://github.com/micropython/micropython/pull/5275 changes the behaviour of `pend_throw()` which is a micropython extension. I agree that behaving like CPython is usually desirable but it can be achieved in a number of ways (one of which does not require any change to `pend_throw()` like in the workaround I posted earlier in this discussion). My points are:

1. Before changing `pend_throw()` I think we should understand if its limitations are deliberate, and if they are deliberate, why they are in place.
2. Are there any downsides in changing `pend_throw()`? e.g. increased code size, unexpected side-effects to existing code or something else.

I like https://github.com/micropython/micropython/pull/5275 too but IMO `pend_throw()` is not broken as-is so it doesn't _need_ to be changed unless we understand the implications.

kevinkk525, 2019-10-29:
Can't say I disagree.
pend_throw seems to be there for injecting an Exception which is not the case when cancelling a not-started generator.

And as uasyncio needs some fixes for cancellation anyway, pend_throw actually doesn't need to be changed (but I may not understand everything to make that assumption):
1. coro gets cancelled by uasyncio.cancel(coro)
2. try pend_throw on coro
  2.1. pend_throw fails because generator is not started
  2.2. search for coro in runq and remove it (requires changes to deque() though), otherwise throw synchronous exception which will make generator yield StopIteration (no changes to deque or pend_throw required)
3. coro not found in runq, search for the coro in waitq
3.1. coro found in waitq, remove from waitq and put on runq
4. coro gets scheduled in runq and handles Exception immediately
5. coro removed and no changes needed to pend_throw, only to utimeq (and optionall deque)

kevinkk525, 2019-10-29:
So basically:

```
def cancel(coro, with_change_to_deque):
    try:
        coro.pend_throw(CancelledError)
        pend = True
    except TypeError:
        pend=False
    if not pend:
        # *should* be in runq if user has not passed a coro that hasn't been added
        if with_change_to_deque:
            loop.runq.remove(coro)
            return
        else:
            try:
                coro.throw(MySyncException)
            except MySyncException:
                pass # will get removed from runq on next execution with StopIteration
    else: # generator executed at least once
        ind=loop.waitq.find(coro) # or similar, not implemented
        if ind: # on runq
            next(coro) # or put it on runq to let it finish there
            loop.waitq.remove(coro) # not implemented
            return
        else: # coro on runq
            if with_change_to_deque:
                loop.runq.remove(coro)
            return
```

jimmo, 2019-10-30:
@dxxb My understanding is that the reason it was implemented this way was to save a few bytes on each generator instance -- it reuses the function's code state stack space to stash the exception. However this isn't possible if the function isn't started. In #5275 I realised that I could tri-state the field that's used to mark whether the function is currently running (it did so by storing the globals, but this wasn't actually necessary). However I'm sure that with some effort you could probably find some other spare bit in the code state to track whether the function is currently running. So in theory you could make the argument that we shouldn't make any changes to the pend_throw behavior and instead claim back those unused four bytes.

But, pend_throw is a MicroPython-specific feature. It doesn't have a reference imlementation to follow so it gets to make its own rules. For better or worse, the same is also true to a large extent of uasyncio.

My preference moving forward would be to move micropython-lib/uasyncio into maintenance mode, and instead work towards embracing CPython-compatible asyncio at a lower level in MicroPython and hopefully/possibly start making use of it in more core features (e.g. hardware APIs like DMA). To that end I'd like to see the implementation of this move into the core repository and be written in C (i.e. extmod/modasyncio.c). Potentially taking advantage of the direction indicated in #5025 to simplify implementation and keep code size small.

So back to uasyncio as it stands today. There are three issues:
1. Immediate cancellation (i.e. make sure the coro sees the CancelledError asap).
My understanding is that this can be largely addressed, as suggested above, by a simple change to uasyncio.cancel. i.e. to unconditionally schedule (call_soon) the coroutine after calling pend_throw. This means it'll still sit in the waitq (see #2) but when it does finally make it to the head, it'll just return StopInteration on the next call.

2. Immediate cleanup
Even if we fix #1, then it'll sit in the waitq for a while. Various fixes have been proposed for this (with and without modification to utimeq). @kevinkk525 's does your suggestion works if the coroutine swallows the CancelledError? i.e. how does it get back onto the right queue? Another idea might be to adapt something a bit like what CPython does where it has a threshold to rebuild the waitq to flush cancelled entries (to prevent it growing without bound). This can be done in uasyncio without modification to waitq, if we keep a set of cancelled coros, of with another extension to objgenerator.c to interrogate a generator instance and find out whether it has finished.

3. Behavior of cancellation for unstarted coroutines (i.e. @kevinkk525 said in the context of modifying pend_throw that "CPython prevents the task from handling the exception if the coroutine hasn't started yet so this is the expected behaviour.")
I assume you mean that the existing behavior of pend_throw is the expected behavior? However, with #5275, if you pend_throw an unstarted coroutine, the coroutine never sees the exception. Calling next() on it will throw the exception, but the coroutine still never actually runs any code. So I believe that is compatible with CPython (from the perspective of the coroutine)? Can you provide an example?

Anyway, here's another simple temporary workaround that addresses 1 and 2 above. Could we modify uasyncio.sleep{,_ms} to sleep in smaller intervals. i.e. conceptually:

async def low_latency_sleep_ms(ms):
  while ms > 0:
    await sleep_ms(20)
    ms -= 20

(Or even the application could provide this rather than uasyncio if it cares about 1 and 2).

That puts a maximum bound on cancellation and cleanup at 20ms. (Bigger or smaller values might make sense -- I'm guessing this is used for something like a background timeout on a request, so this allows for a steady state of 50 cancellations/s). I realise that uasyncio is designed for much higher throughput but I'm unsure how many cancellations/s you expect to be able to do.

dxxb, 2019-10-30:
@dpgeorge, any idea why pend_throw() was originally limited to working only with already started generators? Thank you!

dpgeorge, 2019-10-30:
> any idea why pend_throw() was originally limited to working only with already started generators?

See original commit 63644016669c173190c71d39fb33ad9502fe2b0f, also #3380 for the discussion leading to its implementation.  In particular: ".pend_throw() is an internal function not supposed to be used by end users (only by coro schedulers).", and at some point in that discussion was suggested to name it .cancel() to better match CPython's asyncio.

From that commit you can see why it's not possible to pend-throw to a just-started generator: because it's reusing the Python stack to store the pending object and for a just-started generator this Python stack pointer (sp) is not pointing to a valid slot in the stack, so can't be used (maybe the code could use the first slot but it's not clear this will always work).  Also I guess the fact that sending non-None to a just-started generator is not possible lead to a conclusion that a similar restriction on pend-throw is needed.

kevinkk525, 2019-10-30:
Thanks a lot for your long and detailed answer @jimmo !

> To that end I'd like to see the implementation of this move into the core repository and be written in C 

I am really looking forward to this! It would indeed be a huge benefit in performance, RAM demands and probably maintenance.

> Immediate cancellation (i.e. make sure the coro sees the CancelledError asap).

Executing the generator immediately without other workarounds will cause major problems: What if the Exception is swalloed and additional async code is executed? You'd need to push the coro to the waitq (where it already is) and it will also be executed at the original schedule which might be at an unexpected time.
Therefore if you plan to rewrite uasyncio in .c I'd not put more effort into it. The issue has been around for more than a year, a couple month more won't make a difference now.
The immediate cancellation of an unstarted coro is now fixed, so that's great.

> Immediate cleanup

I updated my pseudo-code as I indeed forgot about the possibility that the generator swallows the Exception and executes some more async code. In that case it would need to be put pack onto waitq (or runq if sleep(0)). However that code still requires changes to utimeq for immediate cleanup.

> Behavior of cancellation for unstarted coroutines

Yes, the PR you created behaves exactly as CPython. The unstarted generator never sees the Exception and is not allowed to run any code.

> another simple temporary workaround that addresses 1 and 2 above.

I guess that is a possible workaround. Timing accuracy could decreases because each sleep_ms(20) could take 25ms or even more with blocking calls in between making a call of sleep(2) easily be off by a lot. Not sure how much of an issue it would actually be.
But as before, if you plan on rewriting uasyncio then it's probably fine to just leave it like that. The problem is well known by now and developers relying on fast cancellation have already adapted to the problem and either use their own workarounds or fast_io from Peter Hinch.

peterhinch, 2019-10-30:
I can't say I'm keen on splitting up the sleep time into smaller increments. That strikes me as performance sapping, and it still leaves uncertainty over when coros are actually cancelled; potentially quite a lot of uncertainty depending on the nature of the competing coros.

I think as a minimum any change should ensure that applications can issue:
```python
asyncio.cancel(coro)
await asyncio.sleep_ms(0)
# NOW coro is definitely cancelled
```
Option 1 achieves this and my experience with fast_io is that it works well. The only penalty is that a coro which will never run again sits on timeq. A small RAM penalty but virtually no performance hit. I maintain a set of cancelled coros so when the coro emerges from waitq it is discarded.

If uasyncio subsequently improves to remove the cancelled task from the queue, application code will be unchanged because the guarantee of immediate cancellation is maintained.

@jimmo Under what circumstance do you envisage the waitq growing without bound?

hoihu, 2019-10-30:
> My preference moving forward would be to move micropython-lib/uasyncio into maintenance mode, and instead work towards embracing CPython-compatible asyncio at a lower level in MicroPython and [...]

I was thinking about exactly this for weeks (implementing asyncio in c) and was about to bring up this topic on the forum. It is clearly outside of what I can do though. If you @jimmo  can come up with a solution for this, that would be absolutely awesome. And combined with other things like (non blocking) DMA e.g. for i2c or SPI... that would be a dream.

peterhinch, 2019-10-31:
A C implementation would be awesome but a substantial project. Before this is started it would be good to see a clear definition of objectives. Perhaps the objectives could be published or discussed in an RFC. A few questions on objectives:

In the Pycopy library Paul implemented improvements aimed at network development. The current design as a Python package enables features like **uasyncio.udp** to be implemented by Python coders and included at runtime as optional extensions. Of necessity these extensions access functionality in **usayncio.core**. Would a C solution be capable of supporting Python extensions? If not, would the specific features in Pycopy be implemented in C?

I would also like to know the status of the issues I listed [above](https://github.com/micropython/micropython/issues/5242#issuecomment-547384276). Which (if any) will be addressed in the Python version, and which will be listed for implementation in C?

peterhinch, 2019-11-01:
A further thought if a root and branch evaluation of usayncio is to be done: consider support for low power mode. My repo has experimental support for **pyb.stop()** which provides significant power saving and offers the potential for battery operated applications.

There are two aspects to this. The uasyncio timebase must be replaced by one using the chip's RTC (the **ticks_ms** clock stops in low power mode). This is straightforward but code differs between Pyboard 1.x and Pyboard D; currently I have not considered other platforms.

Secondly a means must be provided to enter stop/lightsleep and to determine the sleep period. My approach is to allow all tasks on runq to run, then enter lightsleep for a period (which may dynamically be changed).

In any C version of uasyncio it would be good if the timebase could be changed at runtime. A runtime change facilitates debugging via USB: if an active USB connection is detected the code uses the standard timebase and does not issue **stop()**.

My solution is not cross-platform and the issue deserves further thought. Perhaps power saving should be more tightly integrated with uasyncio.

[This doc](https://github.com/peterhinch/micropython-async/tree/master/lowpower) describes it and shows current waveforms illustrating the substantial savings which can be achieved.

dpgeorge, 2019-11-04:
The specific issue of not being able to pend-throw into an unstarted generator was fixed in 5578182ec97509dcc37521b78da5ee31b96f6b4e.  But since this issue has lots of good, related discussion I'll reopen for now.

dxxb, 2019-11-04:
5 days ago I replied to https://github.com/micropython/micropython/issues/5242#issuecomment-547751840 with some comments to @jimmo but I see now my comment didn't make it into the discussion (the fact it was posted while my flight was accelerating on a runway and I subsequently set airplane mode may have something to do with it 😅)

With https://github.com/micropython/micropython/pull/5275 merged I hope it is not too late to discuss making further changes. Before #5275 we could .throw() for synchronous cancellation and .pend_throw() for async cancellation while giving the generator the opportunity to catch the exception. After #5275, in case of pend_throw() when the generator hasn't started yet, the generator no longer has the opportunity to handle the exception.

Something I found useful when writing code for uasyncio was being able to catch the exception or run code in the `finally` clause of the generator in case of cancellation. IMO the generator (AKA task) should be able to always react to cancellation in order to: cleanup, ignore it, postpone it, queue another task.

I would propose to modify current behaviour in a way that still allows to `pend_throw()` the exception when the generator hasn't started but the exception is raised upon executing the yield statement. Comments?

kevinkk525, 2019-11-04:
I disagree because it would be different from Cpython behavior.
A just-started generator/asyncio task is not supposed to see the exception if it is canceled before first execution.

dxxb, 2019-11-04:
> I disagree because it would be different from Cpython behavior.

`pend_throw()` is a micropython extension it is not meant to "behave like CPython". One can build the desired CPython behaviour in uasyncio (or whatever else) on top `throw()` and `pend_throw()` even when applying the change I proposed. In fact my proposed change would re-instate a *pre-existing* and *useful* feature removed by #5275.

> A just-started generator/asyncio task is not supposed to see the exception if it is canceled before first execution.

`uasyncio` is different from `asyncio` because of additional stringent requirements and it doesn't follow CPython behaviour in a lot of places _by design_. A separate `asyncio` library can be implemented if full interoperability with CPython's `asyncio` is desired. We are talking about `pend_throw()` not `throw()`, and `pend_throw()`'s previous behaviour allowed for something useful which doesn't work anymore.

Maybe @pfalcon decision to forbid `pend_throw()` when the generator had not started yet was deliberate and intended to set the behaviour of `pend_throw()` clearly apart from `pend()` and to guarantee that exceptions setup by `pend_throw()` be raised from known locations in the execution flow of the generator (i.e. `yield` statements). Having precise and easily identifiable (i.e. explicit) points where execution can be cancelled or suspended is important to help reasoning about concurrency.

> A generator/task that hasn't started, doesn't have anything to clean up so there is no need to try a way to run the exception catching/finally clause.

Above I mentioned _"cleanup, ignore it, postpone it, queue another task"_, no need to clean up if the generator hasn't started yet but what about the rest?

Say a timeout or some event cancels a task that didn't have a chance to run yet and the task should be re-tried before giving up. I want the code cancelling tasks to be generic and know nothing about the tasks it is cancelling, while the task being canceled (i.e. the description of the steps to be carried out) is often the right place to make decisions on "what happens next".

> A generator might run a lot of code before it even gets to a yield although it had been canceled.

This is why I said the consequences of #5275 should have been evaluated before applying it. Before #5275, once the task was handed off to the scheduler, the task was guaranteed to execute until the first `yield` statement before being cancelled making reasoning about cancellation and concurrent execution easier. After #5275 code before the first yield will execute or not depending on *when* cancellation happens. I believe this makes it harder to write predictable async code.

> It might even exit cleanly if it doesn't use yield so that wouldn't work as expected.

`pend_throw()` (until #5275) only scheduled exceptions to be raised exclusively on yield statements so my proposed change would not result in different behaviour compared to "before #5275". If I wrote a task so that it can `return` immediately, when started under some conditions, then I am not interested in catching exceptions under those conditions; I do not see an issue with that.

dxxb, 2019-11-04:
For reference (from https://docs.python.org/3/library/asyncio-task.html#asyncio.Task):

> cancel()
> Request the Task to be cancelled.

> This arranges for a CancelledError exception to be thrown into the wrapped coroutine on the next cycle of the event loop.

> The coroutine then has a chance to clean up or even deny the request by suppressing the exception with a try … … except CancelledError … finally block. Therefore, unlike Future.cancel(), Task.cancel() does not guarantee that the Task will be cancelled, although suppressing cancellation completely is not common and is actively discouraged.

The documentation I quoted above seems to indicate that CPython's implementation (see test here https://github.com/micropython/micropython/issues/5242#issuecomment-544805887) does not follow the documented behaviour while `uasyncio` did before #5275 

Docs say that calling cancel() should arrange `for a CancelledError exception to be thrown into the wrapped coroutine on the next cycle of the event loop` and `The coroutine then has a chance to clean up or even deny the request by suppressing the exception with a try … … except CancelledError … finally block`. But after #5275 the coroutine is no longer able to catch exceptions when cancelled before it starts.

kevinkk525, 2019-11-04:
To make this answer simple:
Micropython aims to be CPython compatible wherever possible and only divert from CPython behaviour when it is too complex, resource demanding or impossible.

So intentionally diverting the behaviour of cancelling a coroutine just because you like it better that way, is not an acceptable reason.
CPython doesn't give a task the chance to catch the exception if it hasn't ran, then so will micropython. There's no valid reason to make it behave differently.

I'm not sure though why you quoted the CPython docs of asyncio.Task because micropython behaves in the same way, forwarding the Exception to the running coro. The docs don't say anything about a task that hasn't even started yet.

dxxb, 2019-11-04:
While I find #5275 cleans up the code (and reduces size!), it breaks existing generators that rely on being able to catch all `pend_throw()` exceptions, it also makes the semantics of task execution worse and is ultimately unrelated to this issue because this issue is about `uasyncio`'s `cancel()` function not the underlying exception injection mechanism.

Please @dpgeorge, can 5578182ec97509dcc37521b78da5ee31b96f6b4e be reverted to avoid breaking existing code pending a new PR that allows to call `pend_throw()` on generators that have not started yet without preventing generators to catch exceptions. Thank you.

> So intentionally diverting the behaviour of cancelling a coroutine just because you like it better that way, is not an acceptable reason.

@kevinkk525 First, I didn't open this issue so I wasn't the one asking for changes; existing behaviour prior to #5275 was perfectly adequate and I showed how to achieve what you wanted by changing cancel() in the uasyncio library (or wrapping it with your own function -> no micropython-lib changes). Second, #5275 breaks code that relies on the previous behaviour which, I argued, is superior to the current one (more on this later). So, if anything, the pre-exiting behaviour was diverted in order to follow a CPython compatibiliy criteria that just doesn't apply to `uasyncio`.

Note that I did not ask to **return** to the orignal behaviour because I "like it better that way", instead I provided a variety of arguments to justify why the previous behaviour is better than the one implemented by #5275 not least the fact that it **breaks** existing code that relies on being able to catch exceptions.

Arguments Summary:

1. `uasyncio` is not `asyncio` and it wasn't meant to follow CPython's behaviour closely (even if it is of course desirable to do so when possible), so saying _CPython does it in a certain way therefore `uasyncio` should be changed to match_ does not carry the same weight in `uasyncio` and it may break existing code and in fact...
1. #5275 breaks code expecting to always be able to catch exceptions.
1. #5275 makes it harder to write correct (and smaller) async code because a task (unlike before) may or may not execute up to the first `yield` statement when cancel() is called. One of the reasons this is important is that in `async`/`await` style concurrency places in the program flow where execution is suspended/resumed or exception are raised should be explicit (i.e. yield, await).
1. CPython's own documetation says cancellation takes effect on the next execution round and the coroutine should be able to catch exceptions from `cancel()`. No mention of differences between started/unstarted tasks. CPython's `asyncio` implementation does not follow its own documentation!
1. This issue is about a potential problem in the `uasyncio` *library* function `Cancel()` not `pend_throw()`. If at all possible the solution should be at that uasyncio pure-python library level not the underlying mechanism and indeed a pure-python solution was presented.
1. Changes to pre-existing `pend_throw()` behaviour were not necessary to resolve this issue as proven by the provided pure-python workaround.
1. `pend_throw()` is a micropython extension so it does not follow CPython specs at all.
1. `pend_throw()` was probably restricted to already started generators for a reason e.g. an  implementation that stores the pending exception that raises the exception from the next `yield` statement being executed would be complicated or increase code size.

> I'm not sure though why you quoted the CPython docs of asyncio.

`asyncio` implementation does not follow its own documentation  because as you put it: [docs] `don't say anything about a task that hasn't even started yet` so the description of the behaviour in CPython's doc applies to all tasks: started or not started. This makes sense because, when the task is handed to the scheduler, we just want to be able to `cancel()` it and trigger _the same behaviour regarless of the fact the the scheduler got around to start it or not_. Which is why you opened this issue in the first place!

kevinkk525, 2019-11-04:
> Please @dpgeorge, can 5578182 be reverted to avoid breaking existing code pending a new PR that allows to call pend_throw() on generators that have not started yet without preventing generators to catch exceptions. Thank you.

I'm not aware of any code that could break but I'd like to continue that discussion in your PR instead of in this issue because it is considered closed.

> Note that I did not ask to return to the orignal behaviour because I "like it better that way", instead I provided a variety of arguments to justify why the previous behaviour is better than the one [...]

In my opinion it doesn't matter if it breaks code that wasn't CPython compliant anyway and "behaviour is better" is very subjective and micropython follows CPython so you might as well take it up with CPython.

Point 3: You should take it up with CPython because ultimately micropython aims to behave like CPython and now it does. No matter if you feel that it makes it harder to write "correct" code.
Point 4: same.. take it up with CPython if you think it works incorrectly.
Point 5: Where's the difference if pend_throw() does the same job as try: pend_throw() except: throw() ?
Point 6: You are right but as pointed out it was still a good PR so why not?
Point 7-8: And? It serves the goal of making the corresponding CPython library behave correctly like CPython, not provide some additional features for you that are CPython incompatible.

> we just want to be able to cancel() it and trigger the same behaviour regarless of the fact the the scheduler got around to start it or not. Which is why you opened this issue in the first place!

No that is not why I opened this issue. I opened this issue because I couldn't prevent a non-started coroutine from running and actually had to let it run until the first yield, which is exactly what you want but exactly the behaviour that was wrong in my opinion (besides the exception from pend_throw) and differently from CPython.

Further discussions please in your PR.

dxxb, 2019-11-04:
> I'm not aware of any code that could break

My code does, for instance.

> I'd like to continue that discussion in your PR instead of in this issue because it is considered closed.

This issue is open and I think discussion as to how to resolve your use case in a way that doesn't break existing code should continue here.

> In my opinion it doesn't matter if it breaks code that wasn't CPython compliant anyway

I'll say it again: `pend_throw()` is a micropython extension one that was **designed** to behave in a specific way. It is an API, APIs are not broken lightly because people have code that depends on them.

To all the rest of your points: `uasyncio` is not compatible with CPython's `asyncio` and was not meant to be. It cannot be made compatible without dropping its main features, that is way it is called `uasyncio` and not `asyncio`. For `asyncio` compatibility a new `asyncio` module needs to be created.

You figured `uasyncio` is the closest there is to `asyncio` but it doesn't quite do what you want (I have been there believe me). So you want to change `uasyncio` to be more like `asyncio` because it's **soooo** close but not quite. But `uasyncio` is the way it is _by design_, the author didn't decide to make it gratuitously incompatible, `uasyncio` is the way it is because it has to run on very memory constrained devices and `asyncio` would not be able to.

I would like to have a CPython compatible `asyncio` available for micropython too but `uasyncio` is what I found and people wrote code that works with it, me included, so incompatible changes to its API should not be made lightly.

> No that is not why I opened this issue. I opened this issue because I couldn't prevent a non-started coroutine from running and actually had to let it run until the first yield, which is exactly [...]

My bad, so you want `uasyncio` to behave differently to how it _legitimately_ does, please see above for reasons why your use case should not break existing code. I thought we established the following workaround does exactly what you wanted without **any** changes to micropython core?

``` python
def cancel(coro):
    try:
        prev = coro.pend_throw(CancelledError())
        if prev is False:
            _event_loop.call_soon(coro)
    except TypeError as e:
        coro.throw(CancelledError())
```

If so, your problem can be solved locally for you without breaking `uasyncio` for other people. Note that the version of `cancel()` I quoted above should not be added to `uasyncio` as-is because it doesn't give tasks a chance to handle exceptions when the task has not started.

We could change the signature of `cancel(coro)` in `uasyncio` by adding a keyword argument `cancel(coro, kill_unstarted=False)` so the application can pass `True` if they want your desired behaviour.

From your https://github.com/micropython/micropython/pull/5288#issuecomment-549519809

> Please provide a code example of how you were able to catch all exceptions before the changes in #5242 because I couldn't think of any.

Previously all exceptions accepted by `pend_throw()` were raised at the location of `yield` statements so all exceptions could be handled by wrapping them in try: except: finally: blocks.

> But judging from your comments, your changes are incompatible to CPython behaviour and therefore shouldn't be implemented. It would make creating code more confusing.

The intent in #5288 is to **restore** the semantics of `pend_throw()` to pre-#5275 state to fix the exception handling issue #5275 introduces without losing the refactoring that was part of #5275. It is fixing a problem and CPython compatibility has nothing to do with it.

kevinkk525, 2019-11-04:
You have yet to post a minimal example of what actually is broken in your code.

Besides that: uasyncio is not a different asyncio library, it is just a smaller version of asyncio and therefore always has the goal that the features that are implemented are as close to CPython asyncio as possible. 

> Previously all exceptions accepted by pend_throw() were raised at the location of yield statements so all exceptions could be handled by wrapping them in try: except: finally: blocks.

Well it still does that. The only thing that changed is that pend_throw doesn't throw an Exception on an unstarted generator. Therefore your code must do something like this to be broken by the changes:
```
def cancel(coro):
    try:
        asyncio.cancel(coro)
    except TypeError:
        await asyncio.sleep(0)
        cancel(coro)
```

So please provide a code example so we can all see what actually breaks.

dxxb, 2019-11-04:
> Besides that: uasyncio is not a different asyncio library, it is just a smaller version of asyncio and therefore always has the goal that the features that are implemented are as close to CPython asyncio as possible.

Years of commit history, issue tracker and forum discussions about `uasyncio` disagree with your ~hubris~ statement. May I suggest their perusal before deciding for others what `uasyncio` is (or is not) and what its goals are?

`uasyncio` can be improved and, as you have seen from the beginning of this discussion, people were happy to provide potential solutions to your problem and volunteered, myself included, to work on improving `uasyncio`. As I tried to explain earlier, when I first started to use `uasyncio` I wanted it to be more like `asyncio` and did not understand why it was so _annoyingly_ and _artificially_ incompatible, then _after a while_ the apprentice learned it was so for a reason.

> The only thing that changed is that pend_throw doesn't throw an Exception on an unstarted generator.

No, it is **not** the only thing that changed. I pointed out repeatedly in my comments that now an exception scheduled using `pend_throw()` in a not-started generator is not raised in the location of a `yield` statement.

> So please provide a code example so we can all see what actually breaks.

For an example one does not need to look further than the test suite for #5275, see https://github.com/micropython/micropython/commit/5578182ec97509dcc37521b78da5ee31b96f6b4e#diff-0854c0e6e091112656606d1ddf368cc2R80

Any task that relies on execution reaching the first yield statement to catch exceptions will behave differently after #5275

``` python
class CancelledError(Exception):
	pass

foo = 0

def gen_cancelled():
	global foo
	for i in range(5):
		try:
 			yield i
		except CancelledError:
 			print('Do something before cancellation')
 			foo = 10

# Cancel started generator
foo = 1
g = gen_cancelled()
next(g)
g.pend_throw(CancelledError())
next(g)
assert foo == 10, foo

# Cancel not-started generator
foo = 2
g = gen_cancelled()
g.pend_throw(CancelledError())
try:
	next(g)
except Exception as e:
	print("raised", repr(e))
# AssertionError because the generator didn't get the chance to catch CancelledError()
assert foo == 10, foo
```

Previously excution would have **always** reached `yield` before allowing a pending exception to be raised simply because `pend_throw()` did not allow to inject an exception unless the `yield` statement had been reached.

peterhinch, 2019-11-05:
It's self-evident that a script crafted to demonstrate the effects of the PR will do so. The point as I see it is that **pend_throw** is not CPython compatible. It is for internal use in uasyncio and not intended for use in applications.

Does the PR break existing code which uses uasyncio's provided cancellation or timeout capabilities? And, if so, does the behaviour with the PR differ from that of CPython?

kevinkk525, 2019-11-05:
I still don't see where this code should have been broken because the line "g.pend_throw(CancelledError())" would have raised a TypeError before and you would have never gotten to the part where the generator executes until the yield.

Do you have actual code that you actually tested to be broken that you can present?

> Previously excution would have always reached yield before allowing a pending exception to be raised simply because pend_throw() did not allow to inject an exception unless the yield statement had been reached.

Yeah but simply because one had no means of cancelling a non-started generator because calling asyncio.cancel() would raise a TypeError. That means your code would rely on ignoring that exception and scheduling a cancel later when the coro has run once. So please provide a real code example..

dxxb, 2019-11-05:
> It's self-evident that a script crafted to demonstrate the effects of the PR will do so.

And yet it shows that now exceptions can escape handling by the generator which is the root of the issue.

>  It is for internal use in uasyncio and not intended for use in applications.

@peterhinch, `pend_throw()` is a python function (no leading `_`) so it is prefectly reasonable to assume it is public. But more importantly previous `pend_throw()` behaviour is required to ensure generators behave correctly when exceptions are injected which is useful beyond `uasyncio`.

> The point as I see it is that pend_throw is not CPython compatible.

As I pointed out countless times by now this is irrelevant because CPython does not expose or ducument `pend_throw()`, and yet...

> And, if so, does the behaviour with the PR differ from that of CPython?

... the behaviour of `cancel()` in CPython (Python 3.7.4) does not agree with CPython's own documentation of `cancel()`. In order to implement the behaviour of `cancel()`, described by CPython's own documentation, the guaranteee that all exceptions injected by `pend_throw()` are handled by the generator is required. This guarantee was present in previous behaviour and has been removed by https://github.com/micropython/micropython/commit/5578182ec97509dcc37521b78da5ee31b96f6b4e

> Does the PR break existing code which uses uasyncio's provided cancellation or timeout capabilities? 

Yes, it breaks tasks (generators) written to assume they will be *always* given a chance to catch cancellation exceptions and the issues it creates may not be immediately obvious because it is timing dependent.

peterhinch, 2019-11-06:
@dxxb I agree: pend_throw should have a leading `_`. It has been [pointed out](https://github.com/micropython/micropython/pull/3380#issuecomment-352271885) that it is an internal function, but the name should provide a further hint.

Using undocumented internal features in applications carries hazards: the functionality may change or the feature may be removed if it is no longer required to support a future version of uasyncio.

dxxb, 2019-11-06:
> @dxxb I agree: pend_throw should have a leading _. It has been pointed out that it is an internal function, but the name should provide a further hint.

I am sorry but the cat is out of the bag. The function has been public for a long time and it makes sense for people to use it and expect it isn't broken.

However, I would be perfectly happy to make changes to existing code I maintain if it made sense moving forward, but the changes we are talking about have a negative impact on the correct functioning  of `uasyncio`.

stinos, 2019-11-06:
> I am sorry but the cat is out of the bag. The function has been public for a long time and it makes sense for people to use it and expect it isn't broken.

Just a sidenote, but I really don't think MicroPython's intent is generally adhering this principle, see e.g. second paragraph of the main README:

> WARNING: this project is in beta stage and is subject to changes of the code-base, including project-wide name changes and API changes.

There's a whole lot of functions in MicroPython, both in Python and C code, which have been publicly available for a long time but still remain implementation details and hence usually are subject to change. So they should be treated as 'you can use it at your own risk'. I understand that's not always ideal, I've experienced this myself a number of times, but I knew on beforehand this was bound to happen.

dxxb, 2019-11-06:
> Just a sidenote, but I really don't think MicroPython's intent is generally adhering this principle, see e.g. second paragraph of the main README

Acknowledged, although I have the feeling that line is may be a leftover from earlier stages of the project. Even considering the `.pend_throw()` subject to changes, my issue with the changes we are talking about is two-fold: one is with process because legitimate concerns were raised and less invasive solutions proposed but the more invasive and breaking change was rushed through (which I now realised may be part of a not-so-well-publicized effort to replace `uasyncio`) and corrective action was not taken after its issues were highlighted, the other is with the fact that these changes (as they stand right now) IMO prevent implementing a well-behaved async scheduler that fits the same memory footprint `uasyncio` was designed for.

dpgeorge, 2021-07-13:
The new uasyncio v3 makes this issue obsolete.

commit 5578182e (2019-11-04): py/objgenerator: Allow pend_throw to an unstarted generator.

Replace the is_running field with a tri-state variable to indicate
running/not-running/pending-exception.

Update tests to cover the various cases.

This allows cancellation in uasyncio even if the coroutine hasn't been
executed yet.  Fixes #5242
closed 2021-07-13 by dpgeorge