micropython/micropython #5243: WDT erroneously enabled ?

sengpw, 2019-10-21:
When running latest version of micropython (esp32spiram-20191011-v1.11-422-g98c2eabaf.bin) on attached code I get following runtime message:

E (2120640) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (2120640) task_wdt:  - IDLE1 (CPU 1)
E (2120640) task_wdt: Tasks currently running:
E (2120640) task_wdt: CPU 0: IDLE0
E (2120640) task_wdt: CPU 1: mp_task

When running code using micropython (esp32-20190529-v1.11.bin) runtime message does not appear. Is the latest version buggy? Or do I have to feed the watchdog? How?
[main.py.zip](https://github.com/micropython/micropython/files/3753048/main.py.zip)

jimmo, 2019-10-21:
This has already been fixed recently, but for some reason the downloads page hasn't updated for 10 days. Hopefully will be updated soon.

dpgeorge, 2019-10-22:
> for some reason the downloads page hasn't updated for 10 days

I'm looking into this...

dpgeorge, 2019-10-22:
Daily builds are back (it was a problem with relative paths not working in a cron script).

closed 2019-10-22 by dpgeorge