micropython/micropython #5244: [ESP32] collections.deque does not take keyword arguments

nevercast, 2019-10-21:
I would like to construct a deque as follows `collections.deque(maxlen=10)`. For this to work, `iterables` must have a default. The default should be an empty tuple or None, see: https://docs.python.org/3/library/collections.html#collections.deque

`maxlen` should also support keyword arguments but would not require a default, i.e not optional, since I don't need arbitrarily (unbounded) length deques.

dpgeorge, 2019-10-22:
> I would like to construct a deque as follows `collections.deque(maxlen=10)`

You can do this (in a CPython compatible way) via `collections.deque((), 10)`.  Is that enough?

nevercast, 2019-10-22:
This is for the sake of porting libraries from CPython. Is it not worth supporting keyword arguments ?

dpgeorge, 2019-10-22:
>  Is it not worth supporting keyword arguments ?

Well, it's just that it costs something (code space and execution time).  There are quite a few constructors in MicroPython that don't allow keyword args, where CPython would.

One way around this is to define your own `collections.py`:
```python
from ucollections import *
_deque = deque
def deque(iterable=(), maxlen=None):
    if maxlen is None:
        raise ValueError
    return _deque(iterable, maxlen)
```

nevercast, 2019-10-22:
This workaround stub is probably sufficient for me.  Is there a large amount of metadata required for keyword args?

dpgeorge, 2019-10-22:
> Is there a large amount of metadata required for keyword args?

There's the cost of the interned strings (eg `iterable` and `maxlen`, if they aren't already used elsewhere) along with the struct defining the allowed args and their defaults.

nevercast, 2019-10-22:
Alright, maybe this issue can be resolved in the future if anyone ever decides making a heavier `deque` implementation with other features. Until then, we don't need to make it heavier for a single convienence, the proxy function will do.

goatchurchprime, 2019-10-31:
See also:  https://github.com/micropython/micropython-lib/issues/291 and  https://github.com/micropython/micropython/issues/4612 about allowing a [] instead of a () for the place-holding empty term.

More to the point, I'd really like to know what this deque object is supposed to be about.  

I was working with [uasyncio.queues](https://github.com/micropython/micropython-lib/blob/master/uasyncio.queues/uasyncio/queues.py) module where the first line "`from collections.deque import deque`" fails, unless you have installed the [collections.deque](https://github.com/micropython/micropython-lib/tree/master/collections.deque) module from the official [micropython-lib](https://github.com/micropython/micropython-lib) repository. 

This provides a completely incompatible version of deque that's under `collections.deque.deque` instead of `collections.deque`, which anyway has disappeared from the [documentation](https://docs.micropython.org/en/v1.9.4/pyboard/library/ucollections.html).

It would be great to have an explanation as to whether the core collections.deque module is in the process of being deprecated, or if the uasyncio.queues ought to be rewritten to depend on it.

dpgeorge, 2019-11-04:
> I'd really like to know what this deque object is supposed to be about.

Its original purpose was for use in uasyncio, to make an efficient run queue.

> It would be great to have an explanation as to whether the core collections.deque module is in the process of being deprecated, or if the uasyncio.queues ought to be rewritten to depend on it.

uasyncio.queues was added long ago and not updated/maintained with recent changes to other components.

The plan is to rewrite uasyncio from scratch.

goatchurchprime, 2019-11-14:
That seems a little extreme for something that works so well for me!  Is this plan written down anywhere yet?

nevercast, 2019-11-14:
I for one look forward to an asyncio rewrite that has signals that can be awaited

dpgeorge, 2019-11-15:
>  Is this plan written down anywhere yet?

Please see #5332 for a rewrite of uasyncio.