micropython/micropython #5250: WBUS-DIP28 Reset pin does not work.

peterhinch, 2019-10-22:
Momentarily grounding `rst` has no effect. Tested on two samples and confirmed by @jimmo [here](https://forum.micropython.org/viewtopic.php?f=20&t=7121).

The schematics appear to show a connection to the Pyboard's `nrst` line.

dpgeorge, 2019-10-23:
> The schematics appear to show a connection to the Pyboard's `nrst` line.

It's not actually connected through.  The schematics have the nRST on the PYBD's WBUS ending in a small circle, so that is not connected to the NRST net.

The idea here is that you can "hot plug" tiles and reset them without resetting the PYBD itself.

peterhinch, 2019-10-23:
OK. Sorry, I missed that.

closed 2019-10-23 by peterhinch

dpgeorge, 2019-10-23:
Maybe this requires some doc improvement?