micropython/micropython #5251: Pyboard D: BOOT0 Pin at 3.3V does not enter DFU mode

robert-hh, 2019-10-22:
Connecting BOOT0 to 3.3V and pushing reset does not enter DFU mode, but it also does not boot regularly.

jimmo, 2019-10-22:
Which USB port are you using? boot0 will put it into the ST bootloader, which only works on the USB port accessible on the wbus28 adaptor.

Can you use mboot's DFU mode instead? Either machine.bootloader() or hold down USR1 and reset.

robert-hh, 2019-10-22:
I'm using the USB port on Pyboard D. Using machine.bootloader() or USR1 switch works, but for convenience I wanted to use the other method. Using USR1 is 'risky', since if you finger slip off that tiny switch too early, the flash may get erased instead.

dpgeorge, 2019-10-22:
The 3.3V pin is not live during reset because it must be enabled via `Pin('EN_3V3').on()`.  Instead use VUSB/VIN (BOOT0 is 5v tolerant).

But as @jimmo says, that'll get into the ST bootloader, not mboot.  The ST bootloader does not allow to update the external SPI flash.  On the SF6 it's ok because FW fits in internal flash, but on SF2 you must use mboot.

`machine.bootloader()` is a good way to get into mboot without using USR.  Otherwise it is possible to build mboot with support for a boot0-like pin, but that effectively loses a pin for general purpose use.

robert-hh, 2019-10-22:
Actually I used VUSB for my test. And it's sad, that not mboot is activated by boot0. Is that a hardware or software property?

closed 2019-10-22 by robert-hh

dpgeorge, 2019-10-22:
> And it's sad, that not mboot is activated by boot0. Is that a hardware or software property?

It's a hardware property, that boot0 is tied to the ST loader (which is a feature, in case mboot gets corrupted you can recover the board by reprogramming it from scratch).