micropython/micropython #5258: range object should support a more efficient "contains" check

mikewadsten, 2019-10-24:
In Python 3.6, the following computation returns practically instantaneously:

```
>>> 9_999_999 in range(10_000_000)
True
```

However, in MicroPython, the same operation takes many _many_ seconds. This is because the `range` type in MicroPython handles the "contains" check by falling back to iteration, meaning the example above will, presumably, call `range_it_iternext` 10 million times.

At a minimum, I would expect an `x in range(...)` operation where the range in question has a "step" of 1 to effectively just be a bounds check. For extra credit, it could do the same with steps other than 1, etc.

mikewadsten, 2019-10-24:
An alternative to changing the behavior of the `range` object, would be to update the optimizer to have the same result.

`compile_for_stmt_optimised_range` converts `for x in range(start, end, step)` into a for-loop which does _not_ use a range object. Could the same be done for `x in range(start, end, step)`?

stinos, 2019-10-24:
I think all the same reasoning from that issue applies here as well, and the same question: is this from actual code? (I kinda hope not, the other issue could be unintentional, whereas this one would be intentionally doing something which is as far as I'm aware not guaranteed to do what you expect). Anyway I think (not sure) this one also would require even more code to fix/workaround.

mikewadsten, 2019-10-24:
My example of `99_999_999 in range(10_000_000)` is not, but I can easily find examples of Python code that I and others in the Python community have written which uses `x in range(maximum)` as opposed to `x >= 0 and x < maximum` (which, for large maximums, would be the most efficient way to do it in MicroPython currently).

> whereas this one would be intentionally doing something which is as far as I'm aware not guaranteed to do what you expect

By "this one", do you mean that you don't think that something like `if x in range(256):` should boil down to `if x >= 0 and x < 256` automatically?

In reality, my concern here isn't so much the _speed_ of doing a range check, it's the inefficiency (in terms of computation as well as RAM/heap) of creating an iterator over the range, and then walking the iterator, when at least for the step=1 case some _very simple_ code would be able to avoid that extra allocation.

```
>>> 20 in range(21)
range_getiter
range_it_iternext
range_it_iternext
range_it_iternext
range_it_iternext
range_it_iternext
range_it_iternext
range_it_iternext
range_it_iternext
range_it_iternext
range_it_iternext
range_it_iternext
range_it_iternext
range_it_iternext
range_it_iternext
range_it_iternext
range_it_iternext
range_it_iternext
range_it_iternext
range_it_iternext
range_it_iternext
range_it_iternext
True
>>> 1 in range(3)
range_getiter
range_it_iternext
range_it_iternext
True
```

mikewadsten, 2019-10-24:
If we decide the memory-usage and speed improvements to be gained by implementing this are not worth the (probably small) increase in code size, then we should add this to the documentation as an acknowledged difference from CPython (http://docs.micropython.org/en/latest/genrst/builtin_types.html).

Note that the efficiency of range membership testing in CPython isn't just an implementation detail, it's documented behavior: https://docs.python.org/3/library/stdtypes.html#range (emphasis mine)

> Changed in version 3.2: Implement the Sequence ABC. Support slicing and negative indices. **Test int objects for membership in constant time instead of iterating through all items.**

peterhinch, 2019-10-25:
You can write
```python
if 0 <= a <= 1000:
    # do something
```
This is fast in execution and and clear in intent.

stinos, 2019-10-25:
> it's documented behavior: https://docs.python.org/3/library/stdtypes.html#range

Aha, my mistake, I searched a bit yesterday but didn't find that. What I meant with 'as far as I'm aware not guaranteed to do what you expect' was that I didn't know this was actually in the standard.

> By "this one", do you mean that you don't think that something like if x in range(256): should boil down to if x >= 0 and x < 256 automatically? 

Gave this some thought and personally I cannot make up my mind whether it should or not: on one hand since I learned it's documented then yes it should. On the other hand I see disadvantages, making me think it's not worth doing this:
- people might confuse it for a bounds check instead of membership test
- it's a bit unexpected (well, to me, because it's one particular case which is different than the rest; I'd rather have the optimizer do it's thing and if it makes code faster, fine, but I'd rather not make assumptions about it)
- unlikely to be in all implementations soon, if ever, meaning code which runs fast on one implementation might not on another
- likewise, relying on it to be fast might not survive refactoring when the `range` is exchanged for another sequence 
- for a bounds check, it expresses the intent less clear than alternatives (especially concerning the bound condition, i.e. you don't need to think about whether upper and lower are inclusive or not)
- since it's only for integers so if the variable type changes this is going to fail: `a = 1; a in range(100)` is `True` but `a = 1.1; a in range(100)` is `False`  (basically same as the first point but now with example)

But agreed that either it should be implemented (if the code size change really is sufficiently small) or else documented as a difference.

mikewadsten, 2019-10-25:
> You can write
> 
> ```python
> if 0 <= a <= 1000:
>     # do something
> ```
> 
> This is fast in execution and and clear in intent.

That's actually a very good point, it completely slipped my mind that `0 <= a < 10000` is valid syntax.

In any case I'll leave it up to @dpgeorge to decide whether we should simply document it as a difference, or if someone should try implementing it to see how code size is impacted. (I might try this weekend if I find time to do so.)

dpgeorge, 2019-10-26:
> In reality, my concern here isn't so much the _speed_ of doing a range check, it's the inefficiency (in terms of computation as well as RAM/heap) of creating an iterator over the range, and then walking the iterator,

Note that, since `range()` only support small integers, there is no RAM allocation during the walking of the iterator, only to create it at the start.

robert-hh, 2019-10-26:
```if 0 <= a <= 1000:
    # do something```
is only valid as replacement for `if a in range(...)` with step-size 1.