micropython/micropython #5259: MBEDTLS in STM32 port.

iabdalkader, 2019-10-24:
I'm trying MBEDTLS in STM32 port (no LWIP) but I get an exception about the socket not being a stream object... I see that read and write are not set in modusocket.c **socket_stream_p**,  adding those functions seems to fix it. I'm wondering why they're not implemented, is this the right place to add those functions?

dpgeorge, 2019-10-24:
> I see that read and write are not set in modusocket.c **socket_stream_p**, adding those functions seems to fix it. I'm wondering why they're not implemented, is this the right place to add those functions?

Correct, they are not set and they do need to be there to get it working with the `ussl` modules.

Yes this is the right place to add these functions.  The reason they are not implemented is because most boards in stm32 use `extmod/modlwip.c` as the socket implementation, which does have stream read/write capability.

iabdalkader, 2019-10-25:
Thanks for confirming that, I've implemented read/write in modusocket.c and been testing ussl with WINC1500 and it's working fine.

closed 2019-10-25 by iabdalkader