micropython/micropython #5261: ESP8266 stuck in network.STAT_CONNECTING although connected

kevinkk525, 2019-10-26:
Somehow between multiple soft resets I was able to get the esp8266 stuck in a state where sta.isconnect() returns False and s.status() returns network.STAT_CONNECTING forever. 
However it was perfectly connected and I could use the webrepl.
Only a disconnect followed by a connect fixed that.

I know that without a reproducable testcase this issue might be worthless, in which case feel free to close it. But maybe someone familiar with the esp8266 network code can see a possibility for this to happen now that he has the information.

dpgeorge, 2019-10-26:
This is most likely an issue with the underlying esp8266 SDK (if it is an issue at all).  What's happenning is that `wifi_station_get_connect_status()` is returning `STATION_CONNECTING`, causing both `sta.isconnected()` to return `False`, and `sta.status()` to return `STAT_CONNECTING`.

It could be that it's in a state where it's trying to renew the DHCP release (hence connected to the AP and able to send/recv Ethernet packets), so that's why it returns `STATION_CONNECTING` instead of `STATION_GOT_IP`.  But it works because the IP address it got from before is still valid.

kevinkk525, 2019-10-27:
Hmm that is interesting. Thanks.
That means a mqtt library might need to work around that to prevent waiting forever.

Please close the issue if you feel like it is answered/investigated enough.