micropython/micropython #5262: How to load dll in up? Does upy has something like ctypes.CDLL of CPython?

fei2020, 2019-10-27:
I have quickly glanced at uctypes module. DLL is not meantioned in it.
Thanks.

MrSurly, 2019-10-27:
Typically, loadable shared libraries (and non-interpreted executables) are not really a thing in the sort of embedded environment that μP is targeting.  The exception being the Unix port, but you might as well just use regular Python in that case.

peterhinch, 2019-10-27:
DLL's are generally compiled for X86. Could any SoC run them? In terms of MicroPython targets, Unix would be the only option.

jimmo, 2019-10-27:
MicroPython supports this on the Unix port via the FFI module (import ffilib). It's somewhat similar to uctypes.LoadLibrary (and relies on uctypes for return values). See #3078 for discussions around the Windows port. And see https://github.com/micropython/micropython-lib/blob/master/re-pcre/re.py for an example that uses it to load libpcre.

But I would imagine you're probably asking about doing this on a microcontroller platform? As already explained above, this wouldn't be possible with an existing .dll file compiled for x86, but if you're able to build it from scratch, then you might be able to generate a .mpy instead.

Since v1.11 the .mpy format has supported native code, making a .mpy file conceptually similar to a .dll (or equivalently a .so on other platforms). See #5083 for recent work by @dpgeorge to make it possible to compile an existing C library into a .mpy file.

closed 2019-10-27 by jimmo

fei2020, 2019-10-31:
Thank you. I will try.