micropython/micropython #5263: ESP32 Uart TX-error using UART.sendbreak()

ESPeranti32, 2019-10-27:
I'm using the UART-class to send messages over serial line.
Any message is terminated with the 'BREAK'-signal.
It works so far sending more then 1 char's.
But in that special case sending only 1 char terminated with the UART.sendbreak() call,
the leading char will be overwritten or destructed.
Also the call UART.wait_tx_done(100) before sending the BREAK-signal doesn't help.
Only the call: time.sleep(0.002) before UART.sendbreak() will help a bit.
Used code:
...
uart1 = UART(1, baudrate=9600)
while True:
    uart1.write('A')
    while uart1.wait_tx_done(100) == False:
        pass
    # !! sleep required for 1Char- and  break-send !!
    time.sleep(0.002)
    uart1.sendbreak()
    time.sleep(1)

Used environment:
	ESP32 on Wipy3.0
See attached plots for the resulting streams.
Can you fix this problem in further releases?
[image]
[image]

dpgeorge, 2019-10-27:
It's looks like you're using the Pycom firmware (because of the use of `UART.wait_tx_done()` which doesn't exist in the firmware in the esp32 port here).  So please report the issue to Pycom.

ESPeranti32, 2019-10-27:
Thanks for that information. 
That call: UART.sendbreak() already uses uart_wait_tx_done(). So an extra call wouldn't help.
But if I send two char's in sequence (without UART.sendbreak()) than it works!!
I'll report this problem to Pycom too.

dpgeorge, 2019-10-28:
Since this is not a problem with the esp32 port here I'll close this issue.

closed 2019-10-28 by dpgeorge