micropython/micropython #5265: network.LAN is missing from the latest official builds.

jacklinquan, 2019-10-28:
An officially built bin file downloaded from http://micropython.org/download#esp32
esp32-20191028-v1.11-498-gf69ef97f2.bin (latest)
```
>>> import network
>>> dir(network)
['__class__', '__init__', '__name__', 'AP_IF', 'AUTH_MAX', 'AUTH_OPEN', 'AUTH_WEP', 'AUTH_WPA2_PSK', 'AUTH_WPA_PSK', 'AUTH_WPA_WPA2_PSK', 'MODE_11B', 'MODE_11G', 'MODE_11N', 'PHY_LAN8720', 'PHY_TLK110', 'STAT_ASSOC_FAIL', 'STAT_BEACON_TIMEOUT', 'STAT_CONNECTING', 'STAT_GOT_IP', 'STAT_HANDSHAKE_TIMEOUT', 'STAT_IDLE', 'STAT_NO_AP_FOUND', 'STAT_WRONG_PASSWORD', 'STA_IF', 'WLAN', 'phy_mode']
```
Is it arranged somewhere else?

jimmo, 2019-10-28:
The nightly builds currently don't include PPP or LAN. This is because these builds now uses the pre-release 4.0 version of the ESP32 IDF, which is not compatible with the MicroPython PPP or LAN implementation.

The v1.11 release (i.e. esp32-20190529-v1.11.bin) includes the PPP and LAN drivers.

@MrSurly Are you still interested in looking into adding support for PPP/LAN for IDF 4.x?

Otherwise if it doesn't get done in time for MicroPython v1.12, we will provide an both IDF 3.3 version (with PPP & LAN) and a 4.0-beta1 version (with BLE).

jacklinquan, 2019-10-28:
OOPS! I happen to need both LAN and BLE on a project.
It's OK for now. I'm using the build by aykevl (esp32-bluetooth.bin base on v1.10) which supports both LAN and BLE. But that's the old BLE interface implemented by aykevl. I just need some simple BLE UART function.

dpgeorge, 2019-10-29:
The page https://micropython.org/download/#esp32 now includes both IDF3 and IDF4 builds (but LAN/PPP and BLE are still not supported together).

petrkr, 2021-02-17:
Hello,
I think since IDF3 was officially dropped from build system and can not be used since around commit 9f035d6bb71e35ff9acd05b65dbc751887f03af2. Would be nice to support LAN in IDF4.x too. 

How I can help with that, what is problem to not support LAN in IDF4.x ? I tested ESP-IDF ethernet and it works, so there is no problem on SDK side.

I would like also switch to 4.1 or 4.2 IDF, but I need ethernet (LAN7820 with GPIO 17 clock out). I can do myself build of micropython, so if there is some possibility to on this feature, you will safe lot of time with study of options.

Thanks

jacklinquan, 2021-02-18:
> Hello,
> I think since IDF3 was officially dropped from build system and can not be used since around commit [9f035d6](https://github.com/micropython/micropython/commit/9f035d6bb71e35ff9acd05b65dbc751887f03af2). Would be nice to support LAN in IDF4.x too.
> 
> How I can help with that, what is problem to not support LAN in IDF4.x ? I tested ESP-IDF ethernet and it works, so there is no problem on SDK side.
> 
> I would like also switch to 4.1 or 4.2 IDF, but I need ethernet (LAN7820 with GPIO 17 clock out). I can do myself build of micropython, so if there is some possibility to on this feature, you will safe lot of time with study of options.
> 
> Thanks

Thanks @petrkr ,
It would be great if IDF 4 builds support LAN and PPP. Otherwise I would be stuck with IDF 3 and the old toolchain.

dpgeorge, 2021-02-18:
> How I can help with that, what is problem to not support LAN in IDF4.x ?

It's just a matter of rewriting the `ports/esp32/network_lan.c` file to work with the new IDF Ethernet API.

petrkr, 2021-02-18:
I look littlebit to implementation and there is littlebit hard-work by rewrite also SDK parts like https://github.com/espressif/esp-idf/blob/v4.2/components/soc/src/esp32/emac_hal.c#L62 because it uses KConfig variables for things which micropython used as variable during PHY init - namely that GPIO17 as Clock reference...

I think it's pointless to hard-compile some settings like for me would work GPIO17 as output, but someone will need GPIO0 as Input... Maybe could be fine to start with some hard-coded stuff so everyone can use KConfig and then make it more and more to config / INIT. 

Probably desired situation is keep old-format constructor

dpgeorge, 2021-02-18:
> because it uses KConfig variables for things which micropython used as variable during PHY init - namely that GPIO17 as Clock reference

Can we just disable the clock in Kconfig and then configure it ourselves before/after calling the IDF init function?  If that's not possible then I'd ask Espressif to add dynamic configuration of the clock pin to the IDF so we can do it properly and dynamically.

petrkr, 2021-02-18:
I think we can write this "from scratch" and do it ourself
https://github.com/espressif/esp-idf/blob/v4.2/components/esp_eth/src/esp_eth_mac_esp32.c#L287
here is called than low level, which uses KConfig variables, so if we will override this `emac_esp32_init` function. But we have to follow any updated/fixes in SDK if we will copy those functions and write them parametrised instead KConfigured

dpgeorge, 2021-02-19:
Thanks for opening that issue on the IDF.  It looks like they will eventually add support for changing the clock mode dynamically.  In the meantime we could provide 2 builds, one for GPIO16 and one for GPIO17...

petrkr, 2021-03-16:
> Thanks for opening that issue on the IDF. It looks like they will eventually add support for changing the clock mode dynamically. In the meantime we could provide 2 builds, one for GPIO16 and one for GPIO17...

I haven't seen GPIO16 as clock source yet.. LAN8720 uses GPIO17 and if you have module with external oscilator (50mhz) then GPIO 0 is used as clock reference input.

So more better will be to have GPIO17_OUT and GPIO0_IN builds. For me I am using GPIO 17, because I can disassemble or disable oscilator.

RichardAndrews23, 2021-03-22:
Please can I add a vote for having at least one version avaialble with LAN support rather than wait for a universal solution. The Waveshare LAN8720 module is very popular with a 50MHz oscillator feeding GPIO0 with a simple modification using GPIO17 to disable the oscillator during boot. I don't have the skills to offer to help - I'm just very grateful for those who do have the skills and put so much effort into the micropython project - thank you.

petrkr, 2021-03-22:
Actually first version will be hard-coded that or that way.. So there is only build option what clock will be used... After ESP-IDF will offer API to do that from code, it have to be rewritten (or included) too.

I hope to get some time tomorrow to check that out

chtinnes, 2021-04-20:
Is there already some solution or PR to get LAN (LAN8720) working again?

closed 2021-06-17 by jacklinquan

dpgeorge, 2021-06-17:
LAN support was added back for IDF v4.x in 48437cec4547a21af73102c1ccdf079f63fff534