micropython/micropython #5266: webrepl: Doesn't work from https site

jimmo, 2019-10-28:
webrepl serves on a unencrypted websocket, which means that it can't be accessed from a page served over https.

The simple workaround is to manually ensure that you use http://micropython.org/webrepl/ instead of https://micropython.org/webrepl/

Some ideas to improve this:
 - Make webrepl.py on the device support wss
 - Load the page from the device instead -- i.e. serve a bare-bones webrepl.html that includes all its resources and content from a remote site (this could be done in ~100 bytes or so... not much more than just a script tag).
 - Make https://micropython.org/webrepl/ redirect to http://micropython.org/webrepl/  (This is a bit backwards...given that we should be doing the exact opposite via things like HSTS these days).

Carglglz, 2020-02-04:
Hi @jimmo 

> Make webrepl.py on the device support wss

I have a working version of this (both WebREPL and File Transfer) for MicroPython (and Python too, to test it)
All the tests I did were with esp32 MicroPython v 1.12 stable version.

I think the changes are minimal so as soon as #5543 is solved Websocket "secure" for WebREPL and file transfers would be available. 
To make it work it needs a key and certificate in the device and at "user" side:
```
webrepl.start(ssl=True) # This will start listening on wss://x.x.x.x:8833

# or to start Normal WebREPL

webrepl.start() # This will start listening on ws://x.x.x.x:8266
```
I don't know how this would work with [https://micropython.org/webrepl/](https://micropython.org/webrepl/) but since its from a web browser I guess it would need a certificate signed by a CA and not a self-signed certificate (as the ones I used) and modify `function connect(url)` to wrap the socket in SSL. (I don't know JS either so just guessing...)

If there is interest in this I could write a post on the forum or do a PR to discuss the code additions and decide the best way to implement it. 👍

jimmo, 2020-02-05:
> If there is interest in this I could write a post on the forum or do a PR to discuss the code additions and decide the best way to implement it. +1

Great! Thanks for looking into this.  Putting a PR together sounds like a good way forward.

I didn't think any changes would be required for https://micropython.org/webrepl/  -- just changing the URL to start with `wss://` should be sufficient. But I assume you already tested that? You're probably right though that the self-signed cert would be a problem.

If you can send a PR then I can test it out and find out.

Carglglz, 2020-02-05:
PR done!
> I didn't think any changes would be required for https://micropython.org/webrepl/ -- just changing the URL to start with wss:// should be sufficient. But I assume you already tested that? You're probably right though that the self-signed cert would be a problem.

It tries to connect but all I get is `mbedtls_ssl_handshake error`, so yes I think It would need some changes to make it work.

Anyway, I included in the PR a Websocket Python class ('pure' python, no external dependencies),
So the changes in webrepl.py can be tested.

The steps would be:
* Generate a ec key and cert with openssl and name them with the unique id of the device :
e.g.
      - key: `openssl ecparam -out SSL_key30aea4233564.pem -name secp256r1 -genkey`
      - cert: `openssl req -new -key SSL_key30aea4233564.pem -x509 -nodes -days 365 -out SSL_certificate30aea4233564.pem`

* Convert key and certificate to '.DER' format:
       - key:`openssl x509 -in SSL_certificate30aea4233564.pem -out SSL_certificate30aea4233564.der -outform DER`
        - cert: `openssl ec -in SSL_key30aea4233564.pem -out SSL_key30aea4233564.der -outform DER
`
 * Put key and cert ('.der') in the device.
* Start wss: `webrepl.start(ssl=True)`
* In Python:
e.g.
```
>>> from websocket_client import BASE_WS_DEVICE
>>> wss_device = BASE_WS_DEVICE('192.168.1.42', 'mypass', ssl=True, init=True) 
>>> wss_device.wr_cmd("print('Hello')")
Hello
>>> wss_device.close_wconn()
```

jimmo, 2022-05-14:
Looks like HSTS is enabled on micropython.org/webrepl/ now, so the workaround to use http isn't available.

PsuFan, 2022-05-14:
Who runs micropython.org? Should be able to turn it off for a page or put it on a subdomain or build micropython to support https ;)

jimmo, 2022-07-20:
> * Load the page from the device instead -- i.e. serve a bare-bones webrepl.html that includes all its resources and content from a remote site (this could be done in ~100 bytes or so... not much more than just a script tag).

See https://github.com/micropython/micropython/pull/8931 for an implementation of this.

PsuFan, 2022-07-20:
Interesting… thanks for the work on this!

dpgeorge, 2022-07-23:
Fixed by 924e55aca18978215209a2ed81a49b0a6bdcaaa7

closed 2022-07-23 by dpgeorge