micropython/micropython #5267: RFC: mboot with signed/encrypted dfu support

andrewleech, 2019-10-28:
I've got a task on my backlog to add signed and/or encrypted bootloader support to mboot, the stm32 bootloader.

Just wondering if anyone else has looked into this, have suggestions etc?

I've been looking at https://github.com/dmitrystu/sboot_stm32 and it appears quite full featured and pretty much ready to go, so my plan so far is to essentially merge this projects' signing/encrypting support into mboot, perhaps learning from their dfu implementation too.

Either that, or take the current mboot build and configuration setup and layer that on top of sboot_stm32 and use it pretty much as is.

For reference: WebUSB could be worth a closer look as well: https://github.com/lupyuen/codal-libopencm3/tree/master/stm32/bootloader

hoihu, 2019-10-28:
We also have the need in our application to support signed FW images, so perhaps we can share resources / ideas?

We are thinking about implementing it in a way that leaves mboot moreless as is and put the "secure check" path within the micropython application code. That would have the advantage that the complex handling of public keys (e.g. expiry dates, update of keys and/or revoking of keys) could be done in micropython rather than the boot loader, leaving mboot lean as it is now.

So the idea would be to place the signed FW image into a predefined flash sector, similar as it's being done over DFU with the external SPI flash banks on the pyboard d. But it could be done via any communication path, e.g. via UART, wifi etc over custom communication protocols.

Then the micropython app has logic to check and verify the image in the location using a set of public keys that is part of the micropython image itself (e.g. a "public_keys.py" file). 

It would then, on request, start mboot with the address pointed to the fw image. mboot would then set a "dirty" flag or similar, flash the image and clear the flag after successful verifying.

The only adaption to the existing mboot code would be to disable direct DFU flashing to the app's flash segments (e.g. the internal flash in a pboard d) - I think.

What do you think about this approach? Would that be a path that suits your application too?

andrewleech, 2019-10-28:
Actually I had been thinking about exposing a DFU usb endpoint in the normal micropython firmware, this is basically how it would need to be used: use the dfu to load firmware into staging space, verify it, then reboot to flash. In this mode of usage mboot itself wouldn't even need to enable USB.

The big requirement here though is having enough flash space for the image file - it basically can only work for boards with an external flash of some kind. That's possibly ok though, many have spiflash and/or sd card (would need to add sdcard driver to mboot). Perhaps limiting secure dfu to boards such as this isn't too bad.

The main application being involved in the process does raise the (less safe) possibility of the application dfu interface being used to reflash the bootloader too which could be a really useful addition.

The linked secure bootloader does seem to support secure dfu without the staging space or the need for main micropython being aware though which is nice, although then it would be forcing end users to build their own copy of the mboot codebase (with their own secure key) which is less ideal.

andrewleech, 2019-10-29:
Thanks to @jimmo for pointing out https://github.com/micropython/micropython/blob/master/ports/stm32/mboot/fwupdate.py

* mboot can already flash a pre-copied fw image
* mboot can already be updated from application code!

So this entire exercise can likely already be done in user python code, as simple as putting together a nice example for others to use regarding using one of the crypto libraries to handle the signing/encrypting support!

hoihu, 2019-10-29:
> fwupdate.py

yes that was also my intent (using fwupgrade.py to instruct the bootloader
to flash).

Thats why I mentioned that mboot can be left as is. And yes - it would
require to expose a signing crypto API to uPy. mbedTLS would be our
favorite😉.

Perhaps there are some slight adaptions necessary to mboot? eg to force it
using a predefined location in flash to check for images. also you would
probably disallow downgrading the fw - so the usr button handling etc must
be disabled/changed too.

Note that you dont have to have the image in external flash, it could also
be within internal flash in a spare place.

andrewleech, 2019-10-29:
I was surprised that internal flash is safe, but assuming you don't change the location in flash of the internal FS then yes, it'll be perfectly safe (as long as its big enough of course).

There's already AES exposed in mpy in the form of https://docs.micropython.org/en/latest/library/ucryptolib.html from either axtls or mbedtls depending on compile args.

Yes there may warrant some extra build configuration defines for mboot to enable/disable certain user functions like the push-button to enter etc.

andrewleech, 2019-10-30:
This big problem with having the application in charge of authorising an update is you're now in trouble if your release & flash a dud version, if it locks up or the flashing fails such that the application can't do its bit, then you've bricked your board.

If the bootloader can do the decrypt all on its own then it's still always there as your backup, if the application fails you can boot into mboot via user button etc to recover.

hoihu, 2019-11-07:
> This big problem with having the application in charge of authorising an update is you're now in trouble if your release & flash a dud version, if it locks up or the flashing fails such that the application can't do its bit, then you've bricked your board

It could be an advantage if the bootloader can check the signing of the firmware by itself, but it also makes the bootloader more complex and thus error prone (e.g. complexity of key handling, key updates, security algos etc). IMO the bootloader should stay as lean as possible, reduced to the minimum functionality, possibly even without any communication interfaces to the outside world.

If the application is buggy and can't work properly, then as a failback you could Instruct the bootloader to flash a last-known good (and verified) image that has been marked as valid. Perhaps via Jumper. In that case, you'll need actually two spare banks for FW images, one for the last-known-good and one for the newest downloaded (that might be buggy).

It's kind of a downgrade option... That might not be allowed if specific security needs must be met.

andrewleech, 2019-11-07:
I've actually started building a bootloader that can handle the encryption itself. It's coming together surprisingly cleanly and simply to use.
I've never written a brickable bootloader in the past and decided I don't want to start now :-)

The alternate option of having multiple dfu images available to fallback on in the event of failure takes significant flash space which a lot of smaller chips simply wont have. Many of them couldn't support even holding one image at all to support fsload in mboot, hence the deisre to continue to allow direct dfu flashing.

With holding dfu images in flash, that then raises the question of should then be in the normal filesystem, which means either restricting the fs to fatfs or also enabling littlefs in mboot (littlefs is on the way into micropython master currently). 
Then you run into issues if your filesystem gets corrupted, you lose your backup image and your application may not run correctly anymore (if you haven't tested failure conditions enough) so may not be able to flash new images.
Either that or it needs a bit of raw flash space allocated to it to hold images, then you're always consuming flash of a unknown required amount.

So, from a direction of encryption handled in the bootloader, I'm working on a prototype using AES CTR encryption. 

I initially started with just trying to include the existing micropython AxTLS based aes functions in the mboot compile but not surprisingly they didn't fit in the 32k currently allocated to mboot.
I then started writing a driver to use the stm32 hardware crypto engine, it's not that hard to use and there's some nice working examples from mbed: https://github.com/ARMmbed/mbed-os/blob/master/features/mbedtls/targets/TARGET_STM/aes_alt.c
But they wouldn't compile for me, took me way to long to find out that my nice big premium stm32f765 chip doesn't have the hw crypto engine. So that was a dead end.
So I'm now working with a copy of TinyAES ( https://github.com/kokke/tiny-AES-c) which is really simple and works quite well, and most importantly fits easily if aes.c itself is always compiled in release mode.

The (symmetric) 16 byte key I've got stored in the top 16 bytes of the 32k mboot flash space. 
This means it can be compiled into the mboot image easily, or if a placeholder of all 0xFF is used it can then be written over by an end user at first run.

As an aside, none of this precludes the scheme you're talking about. I still think it'd be worth adding options to mboot the disable USB interface and hard-code firmware image filenames for fsload, then you could use application based encryption if preferred (allowing rsa or other higher security asymetric key options possibly), accepting the flash space needs and/or brick failure possibilities.

I think both these schemes are quite compatible with each other.

jonnor, 2024-09-19:
Is this still relevant for recent MicroPython releases? There seem to be a number of commits that reference the issue, back in 2021.

andrewleech, 2024-09-23:
Closed by https://github.com/micropython/micropython/pull/5309

closed 2024-09-23 by andrewleech