micropython/micropython #5269: ESP32 - First time building micropython - Help

ShanmugaCA, 2019-10-28:
Hello,
I followed the steps provided in the micropython/micropython. 

While running the make in the ESP32 port, getting the bellow errors, any clue ?
also, i am new to python and its virtual enviornment
++++++++++

Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
Building with ESP IDF v4
Traceback (most recent call last):
  File "<string>", line 1, in <module>
ModuleNotFoundError: No module named 'pyparsing'
** ERROR **
EDP IDF requires pyparsing version less than 2.4
You will need to set up a Python virtual environment with pyparsing 2.3.1
Please see README.md for more information
Makefile:95: *** Incorrect pyparsing version.  Stop.

++++++++++

thanks in adv
Shanmuganathan

amaza, 2019-10-28:
> File "", line 1, in

pip install pyparsing=2.3.1

Check what Python version is using...

Possibly you need
python3 -m pip install pyparsing=2.3.1

BR.

ShanmugaCA, 2019-10-28:
Hi El lun,

Thanks for replying... no use getting the same error after successful installation of the "pyparsing"

amaza, 2019-10-28:
checking makefile is using python3

PYPARSING_VERSION = $(shell python3 -c 'import pyparsing;
print(pyparsing.__version__)')
ifneq ($(PYPARSING_VERSION),2.3.1)
$(info ** ERROR **)
$(info EDP IDF requires pyparsing version less than 2.4)
$(info You will need to set up a Python virtual environment with pyparsing
2.3.1)
$(info Please see README.md for more information)
$(error Incorrect pyparsing version)
endif

sudo python3 -m pip install pyparsing=2.3.1

so check you are ok with python3 -c  'import pyparsing;
print(pyparsing.__version__)'

~/micropython/ports/esp32$ python3 -c 'import pyparsing;
print(pyparsing.__version__)'
2.3.1

If it doesn't work, please uninstall and reinstall
sudo python3 -m pip uninstall pyparsing
sudo python3 -m pip install pyparsing=2.3.1

Best regards.

ShanmugaCA, 2019-10-28:
Hi Amaza,
thanks again,
no use... 

++++ pyparsing installation is ok+++++

pip install pyparsing==2.3.1
Collecting pyparsing==2.3.1
  Using cached https://files.pythonhosted.org/packages/de/0a/001be530836743d8be6c2d85069f46fecf84ac6c18c7f5fb8125ee11d854/pyparsing-2.3.1-py2.py3-none-any.whl
Installing collected packages: pyparsing
Successfully installed pyparsing-2.3.1
++++

but 'make' failed  for ESP32
++++++
micropython/ports/esp32$ make
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
Building with ESP IDF v4
Traceback (most recent call last):
  File "<string>", line 1, in <module>
ModuleNotFoundError: No module named 'pyparsing'
** ERROR **
EDP IDF requires pyparsing version less than 2.4
You will need to set up a Python virtual environment with pyparsing 2.3.1
Please see README.md for more information
Makefile:95: *** Incorrect pyparsing version.  Stop.
++++++

Sorry for the trouble... 

thanks in adv
Shanmuganathan

amaza, 2019-10-28:
Please check the output of:
python3 -c 'import pyparsing; print(pyparsing.__version__)'

 The pip possibly uses python2.7 instead of python3
so use python3 -m pip install pyparsing=2.3.1

BR

ShanmugaCA, 2019-10-28:
Thanks again!

* The O/p of python3 -c 'import pyparsing; print(pyparsing.__version__)'

+++
Traceback (most recent call last):
  File "<string>", line 1, in <module>
ModuleNotFoundError: No module named 'pyparsing'
+++

* The O/p of python3 -m pip install pyparsing=2.3.1
+++
python3 -m pip install pyparsing=2.3.1
/usr/bin/python3: No module named pip
+++

Not sure where i am missing and how to fix 

many thanks in adv
Shanmuganathan

amaza, 2019-10-28:
sudo apt-get install python3-pip
After installing pip, redo the python3 -m pip install pyparsing=2.3.1

BR

ShanmugaCA, 2019-10-28:
Hi!
Yes, the pyparsing is installed successfully! Thanks a lot!

Best regards
Shanmuganathan

jimmo, 2019-10-29:
As the message suggests, this is best done inside a Python virtual environment, so as not to mess with your system Python libraries (and so that you don't conflict with your package manager). As a general rule, it's best to avoid ever running "pip install" outside of a virtual environment (especially don't use sudo).

Assuming your have Python3 and pip installed via your system package manager.

```bash
# Create a virtual environment in the directory "venv".
python3 -m venv venv
# Enter the virtual environment.
source venv/bin/activate
# Install specific pyparsing version into the virtual environment.
pip install pyparsing=2.3.1
```

Then later you can re-enter the virtual environment just using the "source venv/bin/activate" line.

closed 2019-10-29 by jimmo