micropython/micropython #5270: DTSL example - missing on the docs.

beyonlo, 2019-10-28:
Hello.

Are there any example about to use DTLS? I would like to use UDP protocol to communicate, but in the secure mode, like as a TCP socket using TLS. I found on docs just about ussl for TCP sockets.

Thank you.

MrSurly, 2019-10-28:
I don't believe it's implemented.  A while back, I [took a look](https://github.com/micropython/micropython-esp32/issues/188) at doing so, and despite what I said in that thread, I did indeed abandon it.

IIRC, the underlying stack (mbedtls) supports this.

beyonlo, 2019-10-29:
@dpgeorge Any plans to implement DTLS?

Thank you.

dpgeorge, 2019-10-30:
>  Any plans to implement DTLS?

Not in the short term, but in the medium term (~3 months) it could be done if there's enough interest.

dpgeorge, 2019-10-30:
For reference, using DTLS on CPython is possible via https://github.com/rbit/pydtls

dmartauz, 2019-10-30:
I am also interested in DTLS support, however I would prefer it to be done similarly to ucryptolib.aes with encrypt/decrypt methods. The reason is that the output would be used as UDP payload for NB-IoT modem controlled via AT commands which is not bound to lwip.

dpgeorge, 2019-10-31:
> The reason is that the output would be used as UDP payload for NB-IoT modem controlled via AT commands which is not bound to lwip.

Note that the `ssl` module can currently be used stand-alone, ie without lwip/sockets.  It just requires a suitable stream (eg user class derived from `io.IOBase`) to be passed in to `ssl.wrap_socket`  I would envisage a similar thing for DTLS.

Sebbb, 2021-03-12:
> > Any plans to implement DTLS?
> 
> Not in the short term, but in the medium term (~3 months) it could be done if there's enough interest.

I'm interested as well, I'm searching for an option to encrypt/decrypt UDP packets on an ESP32.

The possibility to communicate securely definitely makes sense, also in the IoT/embedded world :)

closed 2025-02-14 by dpgeorge