micropython/micropython #5272: usock: if connect() fails, socket is considered open -> not cleaned up

mirko, 2019-10-28:
Not sure this is expected behaviour - if so I'm happy to create a PR mitigating this issue for umqtt - but at least I'm not the only one having not taken it into account:

I noticed that a failed usock.connect() - at least under certain circumstances (e.g. OSError = 104 (ECONNRESET)) - leaves the usock-object open, hence not going to be garbage-collected, causing a fd/mem leak if (re)tried e.g. until connection succeeds.

It hit me with the umqtt library, where I'm running mqtt_obj.connect() in a loop, until connection is established.
At some point OSError 104 changes into OSError 23.

I mitigated that by the following patch:

```
diff --git a/umqtt.simple/umqtt/simple.py b/umqtt.simple/umqtt/simple.py
index 8216fa5..9019223 100644
--- a/umqtt.simple/umqtt/simple.py
+++ b/umqtt.simple/umqtt/simple.py
@@ -55,7 +55,15 @@ class MQTTClient:
     def connect(self, clean_session=True):
         self.sock = socket.socket()
         addr = socket.getaddrinfo(self.server, self.port)[0][-1]
-        self.sock.connect(addr)
+        try:
+            self.sock.connect(addr)
+        except Exception as exc:
+            try:
+                # when the connect() fails due to e.g. 104 (ICONNRESET), the socket appears to be considered open - so closing it explicitly.
+                # we don't know though which exact error conditions lead to a (un)closed socket - hence wrapping the close() call as well, skip potentially thrown exceptions but instead throw the one from the failed connect()
+                self.sock.close()
+            finally:
+                raise exc
         if self.ssl:
             import ussl
             self.sock = ussl.wrap_socket(self.sock, **self.ssl_params)
```

To me that however looks rather like treating symptoms, hence the ticket and no PR. Tell me if I'm mistaken.

dpgeorge, 2019-10-29:
Which port do you see this on?  (They have different socket implementations.)

mirko, 2019-10-29:
ESP32

dpgeorge, 2019-10-29:
> I noticed that a failed usock.connect() - at least under certain circumstances (e.g. OSError = 104 (ECONNRESET)) - leaves the usock-object open, hence not going to be garbage-collected

That last part is not necessarily true: if the code doesn't keep a reference to the socket object then it will eventually be garbage collected, and upon collection the socket will be closed by the finaliser.  But that won't help reclaim/free the socket until a garbage collection, so it may be that the memory/fd's available for sockets run out before a collection (and no more sockets does not trigger a GC).

It sounds like the general problem here is that "if umqtt fails in any way then the socket is not explicitly closed".  Eg any socket call in umqtt could file, like a read or write.  So it may be that the solution is to actually protect all umqtt calls in the user application code and always close the umqtt socket before doing the next round.

mirko, 2019-10-29:
Just as a side node: I also added a gc.collect() which didn't help (eventually got OSError 23 exceptions).