micropython/micropython #5276: Uasyncio cancelling coroutine takes as long as next scheduled execution

kevinkk525, 2019-10-29:
Since I didn't see any open issues about it and jimmo asked, I though I'd open this long-standing issue:

If you cancel a coroutine, the exception won't be raised before the coroutine is scheduled next. This is because it just stays in waitq until it would be executed regularly. 
So a coroutine yielding an await uasyncio.sleep(10) will only receive the cancel exception after those 10 seconds have passed.

This results in very late exception handling inside the coroutine but also in canceled coroutines being "stuck" in waitq so the user code has to be careful not to cause a queue overflow if he uses cancellations often.

The fix however requires more than a change to uasyncio, it requires changes to utimeq too so a coroutine can be removed from the queue.
As I understand it Paul has that fixed on his fork.

dpgeorge, 2019-10-29:
Thanks for the explanation.

Do you know how CPython/asyncio handles this case?

kevinkk525, 2019-10-29:
CPython executes the coroutine immediately after cancellation and raises the exception.

peterhinch, 2019-10-29:
Paul handles this by a change to utimeq which enables entries to be extracted out of order. I think this is the best way to do it.

My fork does what may be the best that can be done under official firmware: it issues pend_throw and runs the coro immediately. When the coro emerges from the timeq it is discarded. This is not RAM efficient.