micropython/micropython #5278: There is something wrong with network firmware for pyb v1.1

CaptainJackey, 2019-10-29:
I use the pybv11-network-20190529-v1.11.dfu from www.micropython.org or compile by myself's network firmware for w5500, there is  some problems.

the network.active()  and network.ifconfig('dhcp') both can not use. 

Can someone test for me ?  Thanks.

CaptainJackey, 2019-10-29:
It seems someone else meet this problem:

https://forum.micropython.org/viewtopic.php?f=6&t=7077

robert-hh, 2019-10-29:
network.active() etc. is available in builds with lwip enabled. This is the default for PYBD, but not for PYBV11. if you build the firmware yourself, then you can enable it. For the standard build, network.active() is not required, as far as I remember my test.

CaptainJackey, 2019-10-29:
> network.active() etc. is available in builds with lwip enabled. This is the default for PYBD, but not for PYBV11. if you build the firmware yourself, then you can enable w.g. with:
> `make BOARD=PYBV11 MICROPY_PY_LWIP=1`
> or by adding it to the board specific makefile. For the standard build, network.active() is not required, as far as I remember my test.

I used this command:

`make BOARD=PYBV11 MICROPY_PY_WIZNET5K=5500 MICROPY_PY_LWIP=1`

I was already make the mpy-cross, and still has this error log: 

pi@raspberrypi:~/github/micropython/ports/stm32 $ make BOARD=PYBV11 MICROPY_PY_WIZNET5K=5500 MICROPY_PY_LWIP=1
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
CC ../../extmod/modlwip.c
In file included from ../../py/objlist.h:29:0,
                 from ../../extmod/modlwip.c:32:
../../extmod/modlwip.c:1545:19: error: 'MP_QSTR_sendall' undeclared here (not in a function)
     { MP_ROM_QSTR(MP_QSTR_sendall), MP_ROM_PTR(&lwip_socket_sendall_obj) },
                   ^
../../py/obj.h:92:56: note: in definition of macro 'MP_OBJ_NEW_QSTR'
 #define MP_OBJ_NEW_QSTR(qst) ((mp_obj_t)((((mp_uint_t)(qst)) << 2) | 2))
                                                        ^
../../extmod/modlwip.c:1545:7: note: in expansion of macro 'MP_ROM_QSTR'
     { MP_ROM_QSTR(MP_QSTR_sendall), MP_ROM_PTR(&lwip_socket_sendall_obj) },
       ^
../../extmod/modlwip.c:1549:19: error: 'MP_QSTR_makefile' undeclared here (not in a function)
     { MP_ROM_QSTR(MP_QSTR_makefile), MP_ROM_PTR(&lwip_socket_makefile_obj) },
                   ^
../../py/obj.h:92:56: note: in definition of macro 'MP_OBJ_NEW_QSTR'
 #define MP_OBJ_NEW_QSTR(qst) ((mp_obj_t)((((mp_uint_t)(qst)) << 2) | 2))
                                                        ^
../../extmod/modlwip.c:1549:7: note: in expansion of macro 'MP_ROM_QSTR'
     { MP_ROM_QSTR(MP_QSTR_makefile), MP_ROM_PTR(&lwip_socket_makefile_obj) },
       ^
.........

robert-hh, 2019-10-29:
did you run
make submodules
Then do:
make BOARD=PYBV11 clean
make BOARD=PYBV11 MICROPY_PY_WIZNET5K=5500 MICROPY_PY_LWIP=1

closed 2019-10-29 by CaptainJackey

reverbrick, 2021-08-20:
> did you run
> make submodules
> Then do:
> make BOARD=PYBV11 clean
> make BOARD=PYBV11 MICROPY_PY_WIZNET5K=5500 MICROPY_PY_LWIP=1

A life saver, thanks.