micropython/micropython #5281: How to reset the whole context of upy?

fei2020, 2019-10-31:
I integreted upy into my project(Thanks for your great work). upy works well in my project process.
In one running process, the python script file(A in short) could be reload when it was modified. But another python script file(B in short) that imported by A could be reload when it was modified. Therefore I want to reset "the contex" of upy.

I have tried to use mp_deinit and then re-initialize upy as the same as I do when the process startup. But when I invoke a python script again, MP_STACK_CHECK(mp_stack_check) will fail.

Thank you.

closed 2019-10-31 by dpgeorge

fei2020, 2019-11-10:
I have moved this question to forum.micropython.org(Sorry, I am wondering which forum I should post exactly, "General Discussion and Questions" or "Development of MicroPython").

BUT after several days of waiting, there was no reply yet~

peterhinch, 2019-11-11:
@dpgeorge The query relates to resetting the context in a C module. This may be why the forum queries have elicited no response.

fei2020, 2019-11-14:
@dpgeorge 
I can't get response from forum.micropython.org.
So can you re-open this issue? thank you