micropython/micropython #5284: Round() not Rounding

diginfo, 2019-11-01:
Tried this on Pyboard-D SF6

```
>>> round(65.50000000000001,1)
65.50000000000001
>>>
```

peterhinch, 2019-11-01:
Also on the Unix build. I think this is a regression.
```python
>>> round(65.51, 1)
65.50000000000001
```

peterhinch, 2019-11-01:
The Unix build is similar. Rounding does occur but we also see a consequence of imprecision in the binary representation of floats. Consider:
```python
>>> round(65.506, 2)
65.51000000000001
```
Rounding to 2 places is taking place but the result has a 1 digit error in the 14th decimal place.

yur-spiridonov, 2019-11-02:
The number 65.506 is correctly rounded to 65.51. But 65.51 is not a representable number. Therefore, in a double, this number is represented as 6.55100000000000051159076974727E1 = 6.551000000000001

dpgeorge, 2019-11-04:
This seems to be a problem with printing rather than round itself.  Eg:
```python
import array
x=65.50000000000001
bytearray(array.array('d', [x]))
bytearray(array.array('d', [round(x, 1)]))
```
Output on MicroPython unix and CPython is the same:
```
bytearray(b'\x01\x00\x00\x00\x00`P@')
bytearray(b'\x00\x00\x00\x00\x00`P@')
```

dpgeorge, 2019-11-04:
> This seems to be a problem with printing rather than round itself.

A simple way to show this is:
```python
import array
array.array('d', b'\x00\x00\x00\x00\x00`P@')[0]
```
Output on CPython is `65.5`, on MicroPython is `65.50000000000001`

dpgeorge, 2019-11-04:
This looks to be another case (duplicate of) #4212

yur-spiridonov, 2019-11-04:
> The output on CPython is 65.5, on MicroPython it is 65.50000000000001

It’s just that CPython before rounding out the string 65,50000000000001 rounds up to 15 significant digits (65,5000000000000), and MicroPython, in your case, outputs 16 decimal digits (65,50000000000001).

yur-spiridonov, 2019-11-04:
The number 65.50000000000001  in the internal extended format will be represented by the binary code 1000001.1000000000000000000000000000000000000000000000|101. If this number is rounded to the low bit, then we get 1000001.1000000000000000000000000000000000000000000001=
65.5000000000000142108547152020037174224853515625.
The same can be seen on the number 65.506
 If 16 significant decimal digits are displayed in a line, then we get 65.5000000000000.
If rounding is performed without Guard Digits, truncating all bits after the 53rd, then we will have 1000001.1000000000000000000000000000000000000000000000.  If we print 16 significant digits, we get the number 65.5.  . Maybe MicroPython and CPython can round binary numbers differently.

dpgeorge, 2022-07-26:
As of f9cbe6bc47dd4f5b8e85178caecd6f0de22b4c34, the original case of `round(65.50000000000001,1)
` now prints correctly as 65.5 on PYBD-SF6.

closed 2022-07-26 by dpgeorge