micropython/micropython #5285: Why is frozen.o in BUILD/BUILD

mcr, 2019-11-03:
py/py.mk, line 206 has:

    PY_O += $(BUILD)/$(BUILD)/frozen.o

Why is $(BUILD) repeated twice?  For in-tree builds this puts things in build-GENERIC/build-GENERIC/.
If overriding the target directory with make -C ... BUILD=/my/local/directory  then it repeats a big chunk.  This isn't really a problem, but it has a smell to it.

dpgeorge, 2019-11-03:
Because `frozen.c` is generated and placed in `$(BUILD)/frozen.c`, and there's a general make rule to build `$(BUILD)/*.o` from `*.c`.

mcr, 2019-11-03:
I will perhaps suggest some alternatives.  But, at least now I understand why that was done.  What's I'd like to do is to build firmware out-of-tree, where I have keep the python files that I intend to use.  So far it mostly works, with BUILD=/some/other directory, but it seems to include far more modules than needed and the linker runs out of bss space.

closed 2019-11-03 by mcr