micropython/micropython #5286: Trying to add ethernet support to STM32F407VE

untitledlt, 2019-11-03:
#### Thinks I have done:

- defined `MICROPY_HW_ETH_*` variables in `mpconfigboard.h`
- added `MICROPY_PY_LWIP = 1` to `mpconfigboard.mk`
- added `ETH_*` pins to `pins.csv`

Here is my modified board: https://github.com/untitledlt/BLACK_F407VE

#### Now when I'm trying to build it, i get:
```
In file included from eth.c:31:
eth.c: In function 'eth_mac_init':
eth.c:163:106: error: implicit declaration of function 'STATIC_AF_ETH_RMII_REF_CLK'; did you mean 'STATIC_AF_ETH_RMII__REF_CLK'? [-Werror=implicit-function-declaration]
     mp_hal_pin_config_alt_static(MICROPY_HW_ETH_RMII_REF_CLK, MP_HAL_PIN_MODE_ALT, MP_HAL_PIN_PULL_NONE, STATIC_AF_ETH_RMII_REF_CLK);
                                                                                                          ^~~~~~~~~~~~~~~~~~~~~~~~~~
pin_static_af.h:42:48: note: in definition of macro 'mp_hal_pin_config_alt_static'
         mp_hal_pin_config(pin_obj, mode, pull, fn_type(pin_obj)) /* Overflow Error => alt func not found */
                                                ^~~~~~~
cc1: all warnings being treated as errors
make: *** [build-BLACK_F407VE/eth.o] Error 1
```

What am I doing wrong here?

mcauser, 2019-11-08:
Following on from discussion: https://github.com/mcauser/BLACK_F407VE/issues/6

The STM32F407VET6 does not support ports F,G,H,I.
The LQFP100 pinout only supports ports A,B,C,D,E.
The ZGT6 version of this board has ports A-G.

It's throwing at `STATIC_AF_ETH_RMII_REF_CLK`
According to the F407VE datasheet ETH_RMII_REF_CLK is on `PA1` (page 46).

The F405 and F407 share the same af csv.

In `ports/stm32/boards/stm32f405_af.csv` line 4 has:
PortA,PA1,,TIM2_CH2,TIM5_CH2,,,,,USART2_RTS,UART4_RX,,,ETH_MII_RX_CLK/ETH_RMII__REF_CLK,,,,EVENTOUT,ADC123_IN1

Is it supposed to be two underscores in `ETH_RMII__REF_CLK`, instead of one?

dpgeorge, 2019-11-08:
> Is it supposed to be two underscores in `ETH_RMII__REF_CLK`, instead of one?

It looks like that's a typo in `stm32f405_af.csv`, it should be `ETH_RMII_REF_CLK` to match other boards.

untitledlt, 2019-11-08:
Thanks guys!

closed 2019-11-08 by untitledlt