micropython/micropython #5296: ESP32: two USB devices

poesel, 2019-11-05:
The latest nightly creates two USB devices -  /dev/tty.SLAB_USBtoUART & /dev/tty.usbserial-01A7EA3E. A few weeks ago there was only SLAB_USBtoUART (on Macos).

At least AFAIK. I only noticed this because Pymakr now gets confused and tries to connect through usbserial-01A7EA3E which doesn't work (which is a problem of Pymakr not uPy). Updating the firmware through SLAB_USBtoUART does work though.

dpgeorge, 2019-11-05:
I'm not sure if this is the right place for this report.  There's no USB support on esp32 so there's no way it can appear as two devices at once.  It's the rest of the dev board that will have the USB-UART converter, and that's the thing that has possibly changed.

poesel, 2019-11-06:
Ok, thanks. FWIW the hardware has definitely not changed. But maybe its something on the Mac?
I'll have to investigate.

closed 2019-11-06 by poesel