micropython/micropython #5301: uintptr_t not defined in unix/minimal port

earney, 2019-11-06:
Under Ubuntu 19.10 (gcc version 9.2.1 compiled on 20191008), it seems that uintptr_t is not defined in unistd.h.  (maybe I haven't installed a library I should), but when I make the following change to mphalport.h line 26, all compiles just fine..

#include <unistd.h>
#ifndef uintptr_t
#include <stdint.h>
#endif 

Maybe there was a change and uintptr_t was removed from unistd.h ?

earney, 2019-11-06:
just to make it clear, this is for the unix- minimal port.

dpgeorge, 2019-11-07:
Can you please post the output of a failed build, to see exactly where/what the error is?

earney, 2019-11-07:
Here you go..

~/projects/micropython/ports/unix$ make minimal V=1
make COPT="-Os -DNDEBUG" CFLAGS_EXTRA='-DMP_CONFIGFILE="<mpconfigport_minimal.h>"' \
    BUILD=build-minimal PROG=micropython_minimal FROZEN_MANIFEST= \
    MICROPY_PY_BTREE=0 MICROPY_PY_FFI=0 MICROPY_PY_SOCKET=0 MICROPY_PY_THREAD=0 \
    MICROPY_PY_TERMIOS=0 MICROPY_PY_USSL=0 \
    MICROPY_USE_READLINE=0
make[1]: Entering directory '/home/bearney/projects/micropython/ports/unix'
python3 ../../py/makeversionhdr.py build-minimal/genhdr/mpversion.h
CC unix_mphal.c
gcc -I. -I../.. -Ibuild-minimal -Wall -Werror -Wpointer-arith -Wuninitialized -std=gnu99 -DUNIX -DFFCONF_H=\"lib/oofatfs/ffconf.h\" -Os -DNDEBUG -DMP_CONFIGFILE="<mpconfigport_minimal.h>" -U _FORTIFY_SOURCE -c -MD -o build-minimal/unix_mphal.o unix_mphal.c
In file included from unix_mphal.c:33:
../../py/mphal.h:38:1: error: unknown type name ‘uintptr_t’; did you mean ‘intptr_t’?
   38 | uintptr_t mp_hal_stdio_poll(uintptr_t poll_flags);
      | ^~~~~~~~~
      | intptr_t
../../py/mphal.h:38:29: error: unknown type name ‘uintptr_t’; did you mean ‘intptr_t’?
   38 | uintptr_t mp_hal_stdio_poll(uintptr_t poll_flags);
      |                             ^~~~~~~~~
      |                             intptr_t
make[1]: *** [../../py/mkrules.mk:47: build-minimal/unix_mphal.o] Error 1
make[1]: Leaving directory '/home/bearney/projects/micropython/ports/unix'
make: *** [Makefile:222: minimal] Error 2

earney, 2019-11-07:
Here is the output of uname -a and gcc --version , if that helps

uname -a
Linux cis-VirtualBox 5.3.0-19-generic #20-Ubuntu SMP Fri Oct 18 09:04:39 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux

~/projects/micropython/ports/unix$ gcc --version
gcc (Ubuntu 9.2.1-9ubuntu2) 9.2.1 20191008
Copyright (C) 2019 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

dpgeorge, 2019-11-08:
Thanks for the output.  This issue should have already been fixed by commit 4847460232764a116b731da077074818e316fb55

Are you trying to compile the latest master?