micropython/micropython #5307: Pyboard-D BLE unsolicited disconnects on handle 65535

diginfo, 2019-11-08:
I have reported this on the forum, but think maybe it is a bug and so reporting it here.

I am getting 'unsolicited' disconnect interrupts with handle 65535, this will happen for sure if a device is connected and you do a soft reboot, it appears that during the restart the device is disconnected but as it does not have a valid handle for the new session, that it will trigger this event. 

But I also get it intermittently without a restart, and immediately following this 65535 disconnect, the next connection will fail.

dpgeorge, 2019-11-11:
Link to forum for reference: https://forum.micropython.org/viewtopic.php?f=2&t=7205#p40986

dpgeorge, 2019-11-11:
I ran some tests repeatedly connecting/disconnecting PYBD with PYBD over BLE (one central, one peripheral), but could not reproduce the problem.

Are you able to provide complete code that reproduces this?  I assume the PYBD is the central.  What is the peripheral?