micropython/micropython #5312: License compatibility with Silicon Labs

osresearch, 2019-11-09:
Silicon Labs publishes headers for their EFM32/EFR32 chips under a license that is almost MIT.  Is this compatible with [Micropython's MIT license](https://github.com/micropython/micropython/blob/master/LICENSE)?  Can these headers be included into a port for their board?

```
 * @section License
 * <b>Copyright 2017 Silicon Laboratories, Inc. http://www.silabs.com</b>
 ******************************************************************************
 *
 * Permission is granted to anyone to use this software for any purpose,
 * including commercial applications, and to alter it and redistribute it
 * freely, subject to the following restrictions:
 *
 * 1. The origin of this software must not be misrepresented; you must not
 *    claim that you wrote the original software.@n
 * 2. Altered source versions must be plainly marked as such, and must not be
 *    misrepresented as being the original software.@n
 * 3. This notice may not be removed or altered from any source distribution.
```

osresearch, 2019-11-09:
The ARM headers are under the actual BSD 3-clause license. Is this compatible as well?
```
/* Copyright (c) 2011 - 2014 ARM LIMITED

   All rights reserved.
   Redistribution and use in source and binary forms, with or without
   modification, are permitted provided that the following conditions are met:
   - Redistributions of source code must retain the above copyright
     notice, this list of conditions and the following disclaimer.
   - Redistributions in binary form must reproduce the above copyright
     notice, this list of conditions and the following disclaimer in the
     documentation and/or other materials provided with the distribution.
   - Neither the name of ARM nor the names of its contributors may be used
     to endorse or promote products derived from this software without
     specific prior written permission.
   *
```

osresearch, 2019-11-11:
Here is my fork of the micropython tree that will run on the EFR32 SOC in Ikea's Tradfri smarthome devices. https://github.com/osresearch/micropython/tree/efm32/ports/efm32

dpgeorge, 2019-11-13:
> Silicon Labs publishes headers for their EFM32/EFR32 chips under a license that is almost BSD 3-clause. Is this compatible with [Micropython's MIT license](https://github.com/micropython/micropython/blob/master/LICENSE)?

This BSD-like license is more restrictive (has more conditions) than MIT, but this project already uses a lot of other components with other licenses (eg mynewt-nimble with Apache 2.0, littlefs with BSD 3-clause, stm32lib with custom license) so it should be ok.  Best as a submodule though, in the `lib/` directory.

> The ARM headers are under the actual BSD 3-clause license. Is this compatible as well?

That is fine, other components already use BSD 3-clause.

In the end it's up to the end-user of MicroPython to ensure they comply with all license requirements for the components they use, whatever they may be.  The aim is to keep the core code pure MIT to make it easier to integrate that part into any product/project/etc.

dpgeorge, 2019-11-13:
> Here is my fork of the micropython tree that will run on the EFR32 SOC in Ikea's Tradfri smarthome devices.

Very nice!  It would be good to have support for these MCUs in this repository, the main thing being that it requires someone with time to be able to maintain it...