micropython/micropython #5313: ESP32: Inline assembler code fails

robert-hh, 2019-11-10:
As test I wrote a very simple inline assembler function. Calling it causes a core dump. A more complicates function  which toggled GPIO lines did that, and the registers were properly set. So it seems to tip over upon returning to the callee. Function:
```
@micropython.asm_xtensa
def test():
    movi(a2, 1)

print(test())
```
Core dump:
```
Guru Meditation Error: Core  1 panic'ed (InstrFetchProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x800dfd84  PS      : 0x00060330  A0      : 0x800dfd84  A1      : 0x3ffd0240  
A2      : 0x00000001  A3      : 0x00000000  A4      : 0x40099cb4  A5      : 0x3ffd033c  
A6      : 0x3ffe437d  A7      : 0x00000001  A8      : 0x800e3f00  A9      : 0x3ffd0220  
A10     : 0x00000000  A11     : 0x00000000  A12     : 0x00000000  A13     : 0x3ffe52a0  
A14     : 0x00000001  A15     : 0x000018ae  SAR     : 0x00000004  EXCCAUSE: 0x00000014  
EXCVADDR: 0x800dfd84  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000000  

ELF file SHA256: 0000000000000000000000000000000000000000000000000000000000000000

Backtrace: 0x400dfd81:0x3ffd0240 0x400dfd81:0x3ffd0260 0x400ed6fd:0x3ffd0280 0x400e4048:0x3ffd0320 0x400dfd81:0x3ffd0380 0x400dfdae:0x3ffd03a0 0x400e0893:0x3ffd03c0 0x400eab5d:0x3ffd0450 0x400eae66:0x3ffd0480 0x400e073a:0x3ffd0560 0x400ed9dc:0x3ffd05a0 0x400e4048:0x3ffd0640 0x400dfd81:0x3ffd06a0 0x400dfdae:0x3ffd06c0 0x401012fc:0x3ffd06e0 0x40101548:0x3ffd0780 0x400f5d44:0x3ffd07c0 0x4009759d:0x3ffd07f0

Rebooting...
```

dpgeorge, 2019-11-11:
Inline assembler on esp32 was never tested.  There needs to be some changes to the inline xtensa emitter to make it work with windowed registers.

robert-hh, 2019-11-11:
OK. Then I leave it silent for a while. It was just a trial without an urgent need. Viper code is good enough.

closed 2019-11-11 by robert-hh

dpgeorge, 2019-11-11:
@robert-hh you may also want to try out #5083, for writing faster modules in C.  It should work well on esp32.  There are lots of examples in that PR now.

robert-hh, 2019-11-11:
Yes, I have followed this PR with interest, allowing to add own C functions wihthout touching the source of the core firmware package. The latter causes always trouble when trying to keep it in sync with upstream.

0wwafa, 2025-01-19:
Any news on this?

robert-hh, 2025-01-19:
No change in state. Maybe not needed that urgently, since native C-Modules can be integrated,