micropython/micropython #5315: [esp32] machine/RTC.datetime() does not set microseconds

nevercast, 2019-11-11:
Unrelated to the other issues with RTC, https://github.com/micropython/micropython/issues/4394 and https://github.com/micropython/micropython/issues/3674. The ESP32 RTC implementation does not support the 8th argument in the datetime tuple. Microseconds. 

When calling `RTC.datetime()` the format is `year, month, day, weekday, hour, minute, second, microseconds`. However, when setting the last component, `microseconds` are ignored. This means the RTC can only be set to the lower() second.

As seen here https://github.com/micropython/micropython/blob/258b1478307c1f9d89ed67309e145b9a55aeaadc/ports/esp32/machine_rtc.c#L105-L106, `tv.tv_usec` is not set.

This is an easy fix and a good improvement to the time functionality on the ESP32. Without it, the time cannot be accurately synchronized with other devices.

MrSurly, 2019-11-11:
@nevercast [Here's a fix](https://github.com/MrSurly/micropython/tree/fix-5315), if you'd care to test.  I'll submit as a PR if it looks good.

nevercast, 2019-11-11:
> @nevercast [Here's a fix](https://github.com/MrSurly/micropython/tree/fix-5315), if you'd care to test. I'll submit as a PR if it looks good.

I already PRd a fix that is waiting to land https://github.com/micropython/micropython/pull/5316

Sorry that you didn't see it

MrSurly, 2019-11-11:
No worries.

nevercast, 2019-11-13:
Fixed in https://github.com/micropython/micropython/commit/82d358510b38b092dac204786c193d8109dcf886 via #5316

closed 2019-11-13 by nevercast