micropython/micropython #5321: Switch pyexec_mode_kind enum to avoid creating a data segment?

osresearch, 2019-11-11:
In a minimal build the only value in the data segment is `pyexec_mode_kind` since it defaults to `PYEXEC_MODE_FRIENDLY_REPL`.  If the enums for this and `PYEXEC_MODE_FRIENDLY_RAW` were switched, then there would be no initialized rw-data to be handled.

Jongy, 2019-11-12:
+1, it makes sense to move items from the data segment to the bss. On the other hand, since the code compares against `PYEXEC_MODE_FRIENDLY_RAW` it's better to have this value `0` for cheaper comparisons.

As long as there are only 2 values, I guess we can change all comparisons to `!= PYEXEC_MODE_FRIENDLY_REPL` and it'd be okay ... :)

dpgeorge, 2019-11-13:
Good idea!

> As long as there are only 2 values, I guess we can change all comparisons to `!= PYEXEC_MODE_FRIENDLY_REPL` and it'd be okay

I think that would be a bit confusing.  Better to actually switch around the if-else statements of the code that uses these enums, but that requires a fair number of changes.  So I'd say just make the enum order change for now.

Jongy, 2019-11-13:
If we save 1 data byte and gain 8 text bytes, wasn't it all for nothing? In the minimal build it's the last data item so why not make this change in an `#ifdef PORT_MINIMAL`?

dpgeorge, 2019-11-14:
> If we save 1 data byte and gain 8 text bytes, wasn't it all for nothing? I

I think it's worth trying, to see what the final tally is (of byte changes).  Then evaluate it.

osresearch, 2019-11-14:
With `PYEXEC_MODE_FRIENDLY_REPL==1`
```
LINK build/firmware.elf
   text	   data	    bss	    dec	    hex	filename
  68128	      4	   2524	  70656	  11400	build/firmware.elf
```

And with them swapped there is a small increase:
```
   text	   data	    bss	    dec	    hex	filename
  68140	      0	   2528	  70668	  1140c	build/firmware.elf
```

However, since the data segment is now empty, we can remove the code in `Reset_Handler()` that does the copy resulting in an overall decrease.  Not sure if we can do this automatically at link time, which would be far safer.

```
   text	   data	    bss	    dec	    hex	filename
  68108	      0	   2528	  70636	  113ec	build/firmware.elf
```

Jongy, 2019-11-14:
> Not sure if we can do this automatically at link time, which would be far safer.

I'm not sure how to do that without relinking, but a fair solution would be to remove the copy loop and place an `ASSERT(_edata - _sdata == 0);` in the ldscript, with an explaining comment.

dpgeorge, 2019-11-14:
Just as a related side-note: `lib/utils/interrupt_char.c` and `extmod/modurandom.c` also contain data, which looks harder to remove.

dpgeorge, 2020-02-04:
Addressed by b51a2c266a50163f41cb4320f9b9f68a91132403

closed 2020-02-04 by dpgeorge