micropython/micropython #5322: Listing class attributes in the defined order

andrewleech, 2019-11-12:
I've got a config file management package I'm trying to port to micropython, where the config file structure is defined as a python class: https://pypi.org/project/structured-config/

Some of the features of this module are dependent on knowing the order and iterating over of attributes defined in a class.

Currently the only way I can find to get attributes listed on a class is with `dir()` 
``` python
>>> bar = None
>>> 
>>> class Structure:
>>> 	first = 1
>>> 	foo = 0x0f
>>> 	bar = 0x0b
>>> 	last = 2
>>> 
>>> print(dir(Structure))
['__class__', '__module__', '__name__', '__qualname__', '__bases__', 'bar', 'first', 'foo', 'last']
```

However, looking at the source for `dir()`, I can see that that it works by iterating over every QSTR defined in micropython and checks the class/object for each one to see if it exists (seems a bit backwards to me) so they get listed in the order they're (ever) defined in micropython
https://github.com/micropython/micropython/blob/master/py/modbuiltins.c#L175

Is there any other way in micropython currently this could be accomplished?

andrewleech, 2019-11-12:
I've come up with one solution to my problem, in the two merge requests above. 
One of them provides the `__dict__` attribute for classes, similar to the existing __dict__ attr on instances.
The other one ensures that the local_dict and returned __dict__ are always ordered, if enabled in build config.

dpgeorge, 2020-06-09:
> The other one ensures that the local_dict and returned **dict** are always ordered, if enabled in build config.

I think there's another, simpler way to do this: when a new class is created use an `OrderedDict` for its attributes rather than a standard `dict`.  It only requires changing the call to `mp_obj_new_dict` in `mp_builtin___build_class__`.  Is that enough?

andrewleech, 2020-06-09:
> I think there's another, simpler way to do this: when a new class is created use an `OrderedDict` for its attributes rather than a standard `dict`. It only requires changing the call to `mp_obj_new_dict` in `mp_builtin___build_class__`. Is that enough?

I've tried
``` C
STATIC mp_obj_t mp_builtin___build_class__(size_t n_args, const mp_obj_t *args) {
    assert(2 <= n_args);

    // set the new classes __locals__ object
    mp_obj_dict_t *old_locals = mp_locals_get();
    mp_obj_t class_locals = mp_obj_new_dict(0);
    // keep the locals ordered
    mp_obj_dict_t *dict = MP_OBJ_TO_PTR(class_locals);
    dict->map.is_ordered = 1;
    mp_locals_set(MP_OBJ_TO_PTR(class_locals));
```
and this did work! 
I get the `cls.__dict__.keys()` in the same order as they're defined in the python file,  so yes this change is cleaner than changing all dict's to ordered globally. 

Surely in pretty much all real-world scenarios the number of attributes on a class are likely to be under the threshold for where the ordered map will be faster than the standard one too.

Note: `dir(cls)` still **does not** give them in order, for the same reason as in the top post, so I do still need the class.__dict__ changeset as well for my use case.

dpgeorge, 2020-06-09:
> and this did work!

Awesome!

> Surely in pretty much all real-world scenarios the number of attributes on a class are likely to be under the threshold for where the ordered map will be faster than the standard one too.

Yes, most likely.  So I'd recommend this change over making all dicts ordered by default.  But still I wouldn't enable this feature by default, at least not before doing benchmarking.

> I do still need the class.**dict** changeset as well for my use case.

Yes, understood.